Genesis-Embodied-AI/genesis-world #535: Error reported when loading the ply file

leooo66, 2025-01-10:
The obj file can load normally, but the ply file will report an error
code：
    target_entity = scene.add_entity(
        gs.morphs.Mesh(
            file="urdf/3763/point_sample/sample-points-all-pts-nor-rgba-10000.ply",
            #file="meshes/axis.obj",
            pos=(0.5, 0, 0),
        ),
        surface=gs.surfaces.Default(color=(1, 0.5, 0.5, 1)),
    )

error：
(genesis) (base) jun@Jun:~/Genesis$ /home/jun/miniconda3/envs/genesis/bin/python /home/jun/Genesis/examples/rigid/control_franka.py
[Genesis] [21:35:04] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [21:35:04] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [21:35:04] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [21:35:04] [INFO] Running on [NVIDIA GeForce RTX 4060 Laptop GPU] with backend gs.cuda. Device memory: 7.72 GB.
[Genesis] [21:35:04] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [21:35:04] [INFO] Scene <05fa8ac> created.
[Genesis] [21:35:04] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <0b6f158>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [21:35:04] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <104be94>, morph: <gs.morphs.MJCF(file='/home/jun/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [21:35:04] [INFO] Adding <gs.RigidEntity>. idx: 2, uid: <3627dbd>, morph: <gs.morphs.Mesh(file='/home/jun/Genesis/genesis/assets/urdf/3763/point_sample/sample-points-all-pts-nor-rgba-10000.ply')>, material: <gs.materials.Rigid>.
Traceback (most recent call last):
  File "/home/jun/Genesis/examples/rigid/control_franka.py", line 132, in <module>
    main()
  File "/home/jun/Genesis/examples/rigid/control_franka.py", line 41, in main
    target_entity = scene.add_entity(
  File "/home/jun/Genesis/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/jun/Genesis/genesis/engine/scene.py", line 359, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
  File "/home/jun/Genesis/genesis/engine/simulator.py", line 134, in _add_entity
    entity = self.rigid_solver.add_entity(self.n_entities, material, morph, surface, visualize_contact)
  File "/home/jun/Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py", line 64, in add_entity
    entity = entity_class(
  File "/home/jun/Genesis/genesis/engine/entities/rigid_entity/rigid_entity.py", line 68, in __init__
    self._load_model()
  File "/home/jun/Genesis/genesis/engine/entities/rigid_entity/rigid_entity.py", line 75, in _load_model
    self._load_mesh(self._morph, self._surface)
  File "/home/jun/Genesis/genesis/engine/entities/rigid_entity/rigid_entity.py", line 246, in _load_mesh
    vmeshes, meshes = mu.parse_visual_and_col_mesh(morph, surface)
  File "/home/jun/Genesis/genesis/utils/mesh.py", line 223, in parse_visual_and_col_mesh
    vms = gs.Mesh.from_morph_surface(morph, surface)
  File "/home/jun/Genesis/genesis/engine/mesh.py", line 313, in from_morph_surface
    meshes = mu.parse_mesh_trimesh(morph.file, morph.group_by_material, morph.scale, surface)
  File "/home/jun/Genesis/genesis/utils/mesh.py", line 286, in parse_mesh_trimesh
    meshes.append(gs.Mesh.from_trimesh(mesh=mesh, scale=scale, surface=surface))
  File "/home/jun/Genesis/genesis/engine/mesh.py", line 215, in from_trimesh
    if mesh.visual.defined:
AttributeError: 'VertexColor' object has no attribute 'defined'

zswang666, 2025-01-12:
I tested with a random ply and it works.
Could you provide a minimal test script along with the ply file?

leooo66, 2025-01-16:
> I tested with a random ply and it works. Could you provide a minimal test script along with the ply file?

import argparse

import numpy as np

import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=True)
    parser.add_argument("-c", "--cpu", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(seed=0, precision="32", logging_level="debug", backend=gs.cpu if args.cpu else gs.gpu)

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            dt=2e-3,
            substeps=10,
        ),
        pbd_options=gs.options.PBDOptions(
            particle_size=1e-2,
        ),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(3.5, 0.0, 2.5),
            camera_lookat=(0.0, 0.0, 0.5),
            camera_fov=40,
        ),
        show_viewer=args.vis,
    )

    ########################## entities ##########################
    frictionless_rigid = gs.materials.Rigid(needs_coup=True, coup_friction=0.0)

    plane = scene.add_entity(
        material=frictionless_rigid,
        morph=gs.morphs.Plane(),
    )

    target_entity = scene.add_entity(
    gs.morphs.Mesh(
        file="urdf/3763/point_sample/sample-points-all-pts-nor-rgba-10000.ply",
        
        pos=(0.5, 0, 0),
    ),
    surface=gs.surfaces.Default(color=(1, 0.5, 0.5, 1)),
    )

    # cloth = scene.add_entity(
    #     material=gs.materials.PBD.Cloth(),
    #     morph=gs.morphs.Mesh(
    #         file="meshes/cloth.obj",
    #         scale=1.0,
    #         pos=(0.5, 0.5, 0.5),
    #         euler=(180.0, 0.0, 0.0),
    #     ),
    #     surface=gs.surfaces.Default(
    #         color=(0.2, 0.4, 0.8, 1.0),
    #     ),
    # )

    ########################## build ##########################
    scene.build()

    horizon = 500

    for i in range(horizon):
        scene.step()

if __name__ == "__main__":
    main()

Kashu7100, 2025-02-05:
@zswang666 This seems to occur when we load the `.plt` file with point clouds. 

Like something created by:
```python
import open3d as o3d

pc_points = np.array(points)  # Nx3 numpy array
pc_colors = np.array(colors) /255  # Nx3 numpy array (RGB)

# Create Open3D point cloud
pcd = o3d.geometry.PointCloud()
pcd.points = o3d.utility.Vector3dVector(pc_points)
pcd.colors = o3d.utility.Vector3dVector(pc_colors) 

o3d.io.write_point_cloud("filtered_point_cloud.ply", pcd)
```

closed 2025-11-27 by duburcqa