Genesis-Embodied-AI/genesis-world #540: Scene viewer cropped on Mac

RoryMB, 2025-01-11:
I ran the [`examples/render_on_macos.py`](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py) script exactly as is, but the viewer only displays the view cropped to the bottom-left quarter of the camera view.

Click-dragging causes the view to rotate about a point in space located in the top right corner of the window, as if the camera is being handled correctly but then only the bottom-left quarter of the render is displayed to the window.

Interestingly, the code in [this comment](https://github.com/Genesis-Embodied-AI/Genesis/issues/36#issuecomment-2556229369) renders the video correctly to the file, yet still has the same cropped view in the interactive window.

[image]

Terminal output:
```
2025-01-11 00:11:24.782 Python[44881:2456432] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/9s/w2q9rx954mv1r72ftr97yzmw0000gn/T/org.python.python.savedState
[Genesis] [00:11:25] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [00:11:25] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [00:11:25] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [00:11:25] [INFO] Running on [Apple M1 Max] with backend gs.cpu. Device memory: 64.00 GB.
[Genesis] [00:11:25] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [00:11:25] [INFO] Scene <0bd8b97> created.
[Genesis] [00:11:25] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <48f9cac>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [00:11:25] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <31e0d2b>, morph: <gs.morphs.MJCF(file='/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [00:11:25] [INFO] Building scene <0bd8b97>...
[W 01/11/25 00:11:26.049 2456432] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/11/25 00:11:26.242 2456432] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/11/25 00:11:27.633 2456432] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [00:11:27] [INFO] Compiling simulation kernels...
[Genesis] [00:11:29] [INFO] Building visualizer...
[Genesis] [00:11:30] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`.
[Genesis] [00:11:30] [INFO] Viewer created. Resolution: 886×664, max_FPS: 60.
[Genesis] [00:11:30] [INFO] Running at 5700.44 FPS.
285.96877343696735 FPS
[Genesis] [00:11:30] [INFO] Running at 837.04 FPS.
48.37441900697768 FPS
[Genesis] [00:11:30] [INFO] Running at 463.18 FPS.
48.34876831390993 FPS
[Genesis] [00:11:30] [INFO] Running at 326.91 FPS.
49.36391775630539 FPS
[Genesis] [00:11:30] [INFO] Running at 260.04 FPS.
52.81234968962087 FPS
UNSUPPORTED (log once): POSSIBLE ISSUE: unit 4 GLD_TEXTURE_INDEX_CUBE_MAP is unloadable and bound to sampler type (Float) - using zero texture because texture unloadable
[Genesis] [00:11:30] [INFO] Running at 53.05 FPS.
3.288768032417971 FPS
2025-01-11 00:11:30.521 Python[44881:2456432] +[IMKClient subclass]: chose IMKClient_Modern
[Genesis] [00:11:30] [INFO] Running at 53.26 FPS.
57.22809076148504 FPS
[Genesis] [00:11:30] [INFO] Running at 53.01 FPS.
48.553052577964024 FPS
2025-01-11 00:11:30.550 Python[44881:2456432] +[IMKInputSession subclass]: chose IMKInputSession_Modern
[Genesis] [00:11:30] [INFO] Running at 52.79 FPS.

... And so on
```

RoryMB, 2025-01-12:
I tracked down the problem to here:
[`genesis/ext/pyrender/renderer.py:L57`](https://github.com/Genesis-Embodied-AI/Genesis/blob/70fbb72c56df5aee24adf8e04821be658c8d1a9a/genesis/ext/pyrender/renderer.py#L57)

Removing this (and leaving dpscale at 1) solves the issue for me.

However, the comment implies that setting dpscale to 2 was done for retina displays, but my Mac _does_ have a retina display (M1 Max, 2021). So, I am not sure what the correct approach would be to solve this problem.

snakajima, 2025-01-14:
I can reproduce this bug on my Mac (retina) as well. It seems that not only *dpscale*, but also function *_resize_image()* is not necessary for screen rendering.

APhanSpAARC, 2025-01-14:
Same issue here on my side M1 Macbook Pro

andrewcwlee, 2025-01-17:
Same issue on a M1 Pro Mac here

kevin-shvodian, 2025-02-15:
Same issue on M3 Mac 17in

duburcqa, 2025-03-22:
Fix issue has been fixed on main branch.

closed 2025-03-22 by duburcqa

NevoMir, 2025-08-01:
I still get the cropped view (with the current version of genesis) on macbook pro, M4 Pro chip

duburcqa, 2025-08-01:
There must be something wrong on your install. I'm using M4 Max and I have no issue.