Genesis-Embodied-AI/genesis-world #544: How to ignore collision between specific robot link pairs?

Riften, 2025-01-12:
For now I'm using `RigidEntity.detect_collision` to do collision checking for motion planning. I wonder if I can disable collision between specific links? 

Some robot may have `srdf` file together with its description which contains  `disable_collisions` elements to disable collision checking between a specified pair of links. How can I do this in genesis?

It seems that I can always do that by filtering the detected collision between corresponding geom indices. But is there any other way that can allow me to indicate this during scene initialization?

bxtbold, 2025-01-14:
It seems like there is not info provided related to disabling some collision pairs in the official docs.

A temporary solution could be manually adding disabled collision pairs in [RigidEntitiy._is_ompl_state_valid(state)](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/engine/entities/rigid_entity/rigid_entity.py#L1413), but of course pairs will be only ignored for planning

```diff
    def _is_ompl_state_valid(self, state):
        self.set_qpos(self._ompl_state_to_tensor(state))
        collision_pairs = self.detect_collision()
+      # disable some collision_pair here
        if len(collision_pairs) > 0:
            return False
        else:
            return True
```

zswang666, 2025-02-12:
Are you referring to actually disable collision across links in physics simulation (like some links can actually penetrates into each other) or just ways to filter those out?

duburcqa, 2025-05-24:
Have a look to mujoco documentation regarding collision detection. You can specify 'contype' & 'conaffinity', or an explicit list of excluded collision pairs in MJCF files. There is no such support on URDF files though. Could be relevant to access this option for URDF too. Do you think this feature would be useful for your application?