Genesis-Embodied-AI/genesis-world #548: 自定义的xml 文件不在仿真环境显示

flyme45, 2025-01-12:
my xml can be displayed in mujoco but not in genesis.
but i can see the  Coordinate axis of every links in genesis
[image]
[image]

this is my xml:
`<mujoco>
    <default>
        <joint type="hinge" frictionloss="0.1" damping='2' />
        <position ctrllimited="true" ctrlrange="-3.14 3.14" forcelimited="true" forcerange="-50 50" kp="10" />
    </default>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512" />
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="512" height="512" mark="cross" markrgb=".8 .8 .8" />

        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true" />
        
        <mesh file="path/to/your/stl/file.stl"/>
    </asset>

    <worldbody>
        <light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5.0" dir="0 0 -1" castshadow="false" />
        <light directional="true" diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1" />

        <geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1" />

        <body name="b0" pos="0 0 1">
            <geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />
            <body name="b1" pos="-0.2 0 -0.2" euler="0 90 0">
                <geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />
                <joint name="j0" type="hinge" pos="-0.2 0 0" axis="0 1 0" />
                <body name="b3" pos="0.2 0 0.2" euler="0 -90 0">
                    <geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />
                    <joint name="j3" type="hinge" pos="-0.2 0 0" axis="0 1 0" />
                </body>
            </body>
            <body name="b2" pos="0.2 0 -0.2" euler="0 90 0">
                <geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />
                <joint name="j2" type="hinge" pos="-0.2 0 0" axis="0 1 0" />
            </body>
        </body>
    </worldbody>

    <contact>
        <exclude name="e0" body1="b0" body2="b1" />
        <exclude name="e1" body1="b0" body2="b2" />
        <exclude name="e2" body1="b3" body2="b2" />
    </contact>

    <equality>
        <connect name="c1" body1="b2" body2="b3" anchor="0.2 0 0" />
    </equality>

    <actuator>
        <position joint="j0" ctrllimited="true" ctrlrange="-0.875 0.875" />
        <position joint="j2" ctrllimited="true" ctrlrange="-0.875 0.875" />
        <position joint="j3" ctrllimited="true" ctrlrange="-0.875 0.875" />
    </actuator>
</mujoco>
`
**and i also used stl , obj in the other xml, also can not show in genesis ，same phenomenon as the picture**

flyme45, 2025-01-12:
who can do me a favor ? I'd appreciate it.

flyme45, 2025-01-12:
I skimmed #164, but I think not doing collision detection is a tricky solution, and I think there would be a better solution?

flyme45, 2025-01-12:
<equality>
      <connect name="passive_DoF" body1="link_1" body2="link_4" anchor="0 0.2 0.004" />
  </equality>
the "connect"sign has the other alternative in genesis?

flyme45, 2025-01-12:
the "connect"sign in mujoco has the other replacement in genesis?

Hippozhibos, 2025-01-15:
我是设置了两套<geom>，一套用于可视化，把 contype 和 conaffinity 都设为0。另一套用于碰撞检测，contype, conaffinity都设为1。Genesis官方给的franka的xml文件也是用了类似设置。虽然不知道为什么，但好像genesis里用于碰撞检测的几何体不能可视化，可视化的几何体不能用于碰撞检测。你可以先试试看

closed 2025-03-15 by zswang666