Genesis-Embodied-AI/genesis-world #549: pyglet.gl.ContextException: Could not create GL context

ikhlasbhojani, 2025-01-12:
(genesis-env) ikhlas@ikhlas:~/Desktop/genesis$ python app.py
[Genesis] [22:43:56] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [22:43:56] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [22:43:56] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [22:43:56] [INFO] Running on [Intel(R) Core(TM) i5-8350U CPU @ 1.70GHz] with backend gs.cpu. Device memory: 23.35 GB.
[Genesis] [22:43:57] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [22:43:57] [INFO] Scene <f52d277> created.
[Genesis] [22:43:57] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <0284af5>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [22:43:57] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <5252ac3>, morph: <gs.morphs.MJCF(file='/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [22:43:58] [INFO] Building scene <f52d277>...
[W 01/12/25 22:43:58.488 20738] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/12/25 22:43:58.841 20738] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/12/25 22:44:01.890 20738] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [22:44:01] [INFO] Compiling simulation kernels...
[Genesis] [22:44:07] [INFO] Building visualizer...
libGL error: MESA-LOADER: failed to open iris: /home/ikhlas/anaconda3/envs/genesis-env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /home/ikhlas/anaconda3/envs/genesis-env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri, suffix _dri)
libGL error: failed to load driver: swrast
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 1138, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/window/xlib/__init__.py", line 167, in __init__
    super().__init__(*args, **kwargs)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/window/__init__.py", line 533, in __init__
    context = config.create_context(gl.current_context)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/gl/xlib.py", line 117, in create_context
    return XlibContext(self, share)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/gl/xlib.py", line 152, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context
^CTraceback (most recent call last):
  File "/home/ikhlas/Desktop/genesis/app.py", line 43, in <module>
    scene.build()
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/engine/scene.py", line 607, in build
    self._visualizer.build()
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/vis/visualizer.py", line 104, in build
    self._viewer.build(self._scene)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/vis/viewer.py", line 69, in build
    self._pyrender_viewer = pyrender.Viewer(
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 393, in __init__
    self._initialized_event.wait()
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/threading.py", line 581, in wait
    signaled = self._cond.wait(timeout)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
[Genesis] [23:00:43] [INFO] 💤 Exiting Genesis and caching compiled kernels...

(genesis-env) ikhlas@ikhlas:~/Desktop/genesis$ python app.py
[Genesis] [23:00:52] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [23:00:52] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [23:00:52] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [23:00:52] [INFO] Running on [Intel(R) Core(TM) i5-8350U CPU @ 1.70GHz] with backend gs.cpu. Device memory: 23.35 GB.
[Genesis] [23:00:52] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [23:00:52] [INFO] Scene <75c75da> created.
[Genesis] [23:00:52] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <1dd5af7>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [23:00:52] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <6f53941>, morph: <gs.morphs.MJCF(file='/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [23:00:53] [INFO] Building scene <75c75da>...
[W 01/12/25 23:00:53.665 21307] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/12/25 23:00:54.020 21307] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/12/25 23:00:57.096 21307] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [23:00:57] [INFO] Compiling simulation kernels...
[Genesis] [23:01:02] [INFO] Building visualizer...
libGL error: MESA-LOADER: failed to open iris: /home/ikhlas/anaconda3/envs/genesis-env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /home/ikhlas/anaconda3/envs/genesis-env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri, suffix _dri)
libGL error: failed to load driver: swrast
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 1138, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/window/xlib/__init__.py", line 167, in __init__
    super().__init__(*args, **kwargs)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/window/__init__.py", line 533, in __init__
    context = config.create_context(gl.current_context)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/gl/xlib.py", line 117, in create_context
    return XlibContext(self, share)
  File "/home/ikhlas/anaconda3/envs/genesis-env/lib/python3.9/site-packages/pyglet/gl/xlib.py", line 152, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context

i got this error when i run genesis code please guide me

liuy616161, 2025-02-21:
#207
“
[Luckykantnayak](https://github.com/Luckykantnayak)
[on Jan 16, 2025](https://github.com/Genesis-Embodied-AI/Genesis/issues/207#issuecomment-2594643979)

For me sudo apt-get install mesa-utils and conda install -c conda-forge libstdcxx-ng within the virtual environment worked.”

that's work for me

duburcqa, 2025-03-12:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/777

closed 2025-03-12 by duburcqa