Genesis-Embodied-AI/genesis-world #562: AttributeError: partially initialized module 'genesis' has no attribute 'init'

manumerous, 2025-01-14:
Unfortunately there seems to be a problem with the installation of genesis resulting in the examples failing to run. I saw that there have already been similar issues raised (#113 #31 ) but I could not find any solution proposed in them. 

To install genesis on my local linux system i did the following steps: 
```
conda create -n genesis python=3.11
conda activate genesis
conda install pytorch torchvision -c pytorch
pip install genesis-world
```
From the terminal output all the steps seem to be successful. When trying to run the example code with active conda environmnet: 

```
import genesis as gs

gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
)

scene.build()

for i in range(1000):
    scene.step()
```
 I get the following error:
```
python genesis.py 
Traceback (most recent call last):
  File "/home/manu/src/genesis_test/genesis.py", line 1, in <module>
    import genesis as gs
  File "/home/manu/src/genesis_test/genesis.py", line 3, in <module>
    gs.init(backend=gs.cpu)
    ^^^^^^^
AttributeError: partially initialized module 'genesis' has no attribute 'init' (most likely due to a circular import)
```

Do you know how to fix this or how to install the "right version" of genesis?

When doing a pip feeze command I can see that genesis world is installed int he following version: `genesis-world==0.2.1`.

JisuHann, 2025-01-16:
Same error exist for docker too.

Kira-Pgr, 2025-01-16:
I solved the issue by renaming `genesis.py` to something else

KafuuChikai, 2025-01-17:
Hi there,

I believe I have solved this problem.

If you are using **conda** to build the Genesis environment, you need to export the path:

```bash
export PYTHONPATH=/path/to/Genesis:$PYTHONPATH

manumerous, 2025-01-17:
Thank you this fixed the error indeed! It might be worth changing the file name in the examples to avoid everyone from running into this error.

closed 2025-01-17 by manumerous

tawfiqmakram, 2025-01-19:
> I solved the issue by renaming `genesis.py` to something else

rename what?

varshneydevansh, 2025-01-28:
Hi  @KafuuChikai 

My friend and I are having trouble from past 3 days while getting this install can you have the look at the images and the output which I have posted

https://github.com/Genesis-Embodied-AI/Genesis/issues/226