Genesis-Embodied-AI/genesis-world #565: trimesh is not defined

laurrrra07, 2025-01-14:
self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
                           ^^^^^^^
NameError: name 'trimesh' is not defined

zswang666, 2025-02-14:
where did you get the error? we are using our own custom trimesh and it should be installed when installing genesis.

hughlv, 2025-03-20:
Any update? I got same error and don't know what to do. It's a Macbook M1 running macOS Sequoia 15.4

[image]

duburcqa, 2025-03-20:
Are you using the pre-compiled binaries available on pypi? Did you try installing Genesis main branch? I should work now!

closed 2025-03-22 by duburcqa