Genesis-Embodied-AI/genesis-world #570: Fixed Hollow Block Collides and Flies Away When Dropped

Richal13Yu, 2025-01-14:
I encountered an issue while using Genesis to simulate a fixed hollow block dropped onto a column. When I attach the hollow block to the column(fixed=True), everything works as expected. However, if I drop the hollow block from a higher position:

- It collides with the column.
- Instead of dropping to the bottom of the column as expected, it flies away unexpectedly.

Richal13Yu, 2025-01-16:
hi, I saw the issue==444, and tried it in my code, but it still fails to convert my nonconvex item.

[image]
here is my code.

board_surface = gs.surfaces.Glass(color=(0.8, 1.0, 0.0),vis_mode='collision')
block_surface = gs.surfaces.Plastic(color=(1.0, 0.0,0.8),vis_mode='collision')

coacd_options=gs.options.CoacdOptions(threshold=0.1, preprocess_resolution=10)

pos_block=np.array([0.033, 0.05075, 0.08])
pos_block[0]-=0.077
pos_block[1]-=0.0515
block = scene.add_entity(

    gs.morphs.Mesh(file='stl/block.stl',
                   pos=pos_block,
                   fixed = True,
                   decompose_nonconvex = 1,
                   coacd_options = coacd_options,
                   ),
    surface = block_surface
)
board = scene.add_entity(
    gs.morphs.Mesh(file='stl/board.stl',
                   fixed = True,
                   pos=[0,0,0.02],
                   decompose_nonconvex=1,
                   coacd_options=coacd_options,
                   ),
    surface = board_surface
)

Richal13Yu, 2025-01-17:
> Do you have a video?

Sure! The block shoud fall in the boards' collomn, but they maybe loaded wrong 

https://github.com/user-attachments/assets/b7216860-e303-4fcd-bc5a-1e3d07186a50

https://github.com/user-attachments/assets/64c84fab-fcd7-4b65-9731-88e2b9eacea0

wangyian-me, 2025-01-18:
First of all, set `preprocess_resolution=100` to better preserve the original geometry.
Secondly, the coacd is not guaranteed to fully preserve geometry even when `preprocess_resolution=100`, you might want to use the sdf based contact in this case. Adjusting the sdf resolution might help.

Richal13Yu, 2025-01-20:
> First of all, set `preprocess_resolution=100` to better preserve the original geometry. Secondly, the coacd is not guaranteed to fully preserve geometry even when `preprocess_resolution=100`, you might want to use the sdf based contact in this case. Adjusting the sdf resolution might help.

Thank you for the suggestion. However, after adjusting both preprocess_resolution and the SDF resolution, the problem still persists. The block still cannot be loaded correctly.

Could you please provide further guidance or suggest other configurations that might help resolve this issue?

Thank you!

wangyian-me, 2025-01-20:
What do you mean by `adjusting both preprocess_resolution and the SDF resolution`? I think it's a parallel choice and can't be done both. 
By the sdf solution, I mean that set `convexify=False and decompose_nonconvex=False`, and turn SDF resolution higher.

wangyian-me, 2025-01-20:
Also, can you provide the collision mesh after setting preprocess_resolution=100?

Richal13Yu, 2025-01-20:
> What do you mean by `adjusting both preprocess_resolution and the SDF resolution`? I think it's a parallel choice and can't be done both. By the sdf solution, I mean that set `convexify=False and decompose_nonconvex=False`, and turn SDF resolution higher.

OH, thank you!
That really works, but the collomn of the board still cannot be loaded.

[image]

wangyian-me, 2025-01-20:
Is this the sdf result?

Richal13Yu, 2025-01-20:
> Is this the sdf result? Can you show the code that renders this image?

Sure!
import genesis as gs
import numpy as np

gs.init(backend=gs.gpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(
    gs.morphs.Plane(),
)

board_surface = gs.surfaces.Glass(color=(0.8, 1.0, 0.0),vis_mode='collision')
block_surface = gs.surfaces.Plastic(color=(1.0, 0.0,0.8),vis_mode='collision')

coacd_options=gs.options.CoacdOptions(threshold=0.1, preprocess_resolution=100)

pos_block=np.array([0.033, 0.05075, 0.1])
pos_block[0]-=0.077
pos_block[1]-=0.0515
block = scene.add_entity(
    gs.morphs.Mesh(file='stl/block.stl',
                   pos=pos_block,
                   fixed = True,
                   convexify= 0,
                   decompose_nonconvex = 0,
                   coacd_options = coacd_options,
                   ),
    surface = block_surface
)

coacd_option_board=gs.options.CoacdOptions(threshold=0.1, preprocess_resolution=100)

board = scene.add_entity(
    gs.morphs.Mesh(file='stl/board.stl',
                   fixed = True,
                   pos=[0,0,0.02],
                   convexify=0,
                   decompose_nonconvex=0,
                   coacd_options=coacd_option_board,
                   ),
    surface = board_surface
)

psm = scene.add_entity(
    gs.morphs.URDF(file='urdf/psm/psm_RL.urdf',
                   fixed=True,
                   collision=True,
                   pos=(0.5, 0, 0.2),
                   euler=(0.0, 0, 90.0),
                   ),
)
scene.build()

wangyian-me, 2025-01-21:
Have you tried to increase the sdf resolution?

wangyian-me, 2025-01-21:
scene.add_entity(material=gs.materials.Rigid(sdf_cell_size=0.0001, sdf_min_res=32, sdf_max_res=384), ...

Richal13Yu, 2025-01-21:
> Have you tried to increase the sdf resolution?

Thank you for the suggestion. Unfortunately, I can only use the current SDF file, and while I agree that a higher resolution might improve the results, I am unable to adjust it further.

> scene.add_entity(material=gs.materials.Rigid(sdf_cell_size=0.0001, sdf_min_res=32, sdf_max_res=384), ...

the following code does not seem to work in my case:

board = scene.add_entity(
    gs.morphs.Mesh(file='stl/board.stl',
                   fixed = True,
                   pos=[0,0,0.02],
                   convexify=0,
                   decompose_nonconvex=0,
                   coacd_options=coacd_option_board,
                   ),
    material=gs.materials.Rigid(
        sdf_cell_size=0.0001,
        sdf_min_res=32,
        sdf_max_res=384
    ),
    surface = board_surface
)

duburcqa, 2025-03-23:
Do you have any self contained reproduction scripts including all the assets ? Otherwise this issue is not really actionable.

closed 2025-05-24 by duburcqa