Genesis-Embodied-AI/genesis-world #576: AttributeError: 'Viewer' object has no attribute '_pyrender_viewer'

YMIR-GAO, 2025-01-15:
My python version is 3.12.8. I tried the viewer case as document and got the following error. Any solution for that? thanks.

cam_pose = scene.viewer.camera_pose()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/gao/.pixi/envs/default/lib/python3.12/site-packages/genesis/vis/viewer.py", line 193, in camera_pose
    return np.array(self._pyrender_viewer._trackball._n_pose)
                    ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Viewer' object has no attribute '_pyrender_viewer'

BJHYZJ, 2025-02-13:
@YMIR-GAO Have you solved the problem?

YilingQiao, 2025-03-08:
Can you install genesis locally by `pip install -e .`?

duburcqa, 2025-03-22:
This issue has been fixed on main branch.

closed 2025-03-22 by duburcqa