Genesis-Embodied-AI/genesis-world #577: trimesh is not defined

formoree, 2025-01-15:
[image]
I just rented a 4090 server with torch and python installed on the server itself, and I pip-installed genesis-world as requested.But I got this error when running the hello_world demo file.

formoree, 2025-01-15:
I found that switching to another cloud server rental platform and using a higher version of Cuda can solve this problem.

formoree, 2025-01-15:
[image]
After trying to run the hello genesis file with no error, I tried to run my own project file.On my local 4060ti computer, each step takes about 4~5 seconds, of which the collection time is about 4 seconds, but it takes 140 seconds here. I don't know why. I have rented multiple cloud servers and all have similar problems.

Kashu7100, 2025-01-15:
Is your simulation using GPU (I suspect it is working with CPU)? Can you check the initial log for the backend?
For your first comment it says backend is `gs.cpu` (although you mentioned you are using 4090 server).

formoree, 2025-01-16:
Thank you very much for your reply. The first hello_world demo is specified as gs.cpu in the file, so it shows as CPU. In the third comment later, I specified it as GPU, but the GPU call is 0, and there is no traditional log output from gs.init either.

formoree, 2025-01-20:
I apologize, but I cannot currently access the details of my cloud server rental, so I am unable to check the versions of CUDA and NVIDIA drivers. However, I recall that the CUDA versions do not exceed 12.2. Moreover, I have consecutively configured four servers, all of which have similar issues.

duburcqa, 2025-05-24:
This issue has been fixed on main branch.