Genesis-Embodied-AI/genesis-world #578: Questiosn regarding robot model creation and dynamics calculation

gabrielpeixoto-cvai, 2025-01-15:
First, thank you for the fantastic job and for sharing this with the community. I have two questions that are crucial for my current simulator use case:

1. Is there a plan to support dynamic robot model creation? (for example, mujoco has [mjspec](https://mujoco.readthedocs.io/en/stable/programming/modeledit.html) that allows me to create a robot programatically)
2. Is there a way to execute forward dynamics and inverse dynamics? Mujoco has these options and I am currently using the to estimate forces in joints.

Thanks in advance for any answers!

Tadinu, 2025-03-09:
@gabrielpeixoto-cvai, for 1.,
I suppose, either starting from MJCF/XML or composing it programmatically in memory using mjSpec, you could then pass the resulting [compiled mjModel](https://mujoco.readthedocs.io/en/stable/python.html#construction) before feeding it to an Entity.
For specifics:
*  Create a [MujocoMorph](https://github.com/Tadinu/Genesis/blob/dev/genesis/options/morphs.py#L487) class that could receive mjModel directly (instead of a MJCF file path).
*  Pass the created MujocoMorph instance to [gs.Scene.add_entity()](https://github.com/Tadinu/Genesis/blob/dev/genesis/system/base_system.py#L37)

Tadinu, 2025-03-09:
@gabrielpeixoto-cvai, for 1.,
I suppose, either starting from MJCF/XML or composing it programmatically in memory using mjSpec, you could then use the resulting [compiled mjModel](https://mujoco.readthedocs.io/en/stable/python.html#construction) to create an entity in scene.
Specifically,
*  Create a [MujocoMorph](https://github.com/Tadinu/Genesis/blob/dev/genesis/options/morphs.py#L487) class that could receive mjModel directly (instead of a MJCF file path).
*  Pass the created MujocoMorph instance to [gs.Scene.add_entity()](https://github.com/Tadinu/Genesis/blob/dev/genesis/system/base_system.py#L37)

gabrielpeixoto-cvai, 2025-03-10:
@Tadinu 

Thanks for the suggestion, I have considered this path, however, that would still require me to use mujoco, and I rather use the minimum amount of tools. For now due to the limitations of genesis to my use cases, I decided to not use it.

duburcqa, 2025-04-20:
There is currently no way to design robot programmatically, except by writing MJCF file programmatically. You can have a look to the various example in `tests/test_rigid_physics.py`