Genesis-Embodied-AI/genesis-world #580: CMake Could NOT find Python when doing cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON

sickopickle, 2025-01-15:
I'm on Ubuntu 22.04.

I've tried completely uninstalling CMake and and reinstalling via `sudo snap install cmake --classic` as well as deleting my Genesis folder and cloning again. Running `which cmake` returns `/snap/bin/cmake`. Running 'which python' returns `/usr/bin/python`. After following the instructions I keep getting the following error. 

```
CMake Error at /snap/cmake/1437/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Python (missing: Python_EXECUTABLE Python_INCLUDE_DIRS
  Python_LIBRARIES Interpreter Development Development.Module
  Development.Embed) (Required is exact version "3.9")

      Reason given by package: 
          Interpreter: Wrong version for the interpreter "/bin/python"

Call Stack (most recent call first):
  /snap/cmake/1437/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /snap/cmake/1437/share/cmake-3.31/Modules/FindPython/Support.cmake:4002 (find_package_handle_standard_args)
  /snap/cmake/1437/share/cmake-3.31/Modules/FindPython.cmake:631 (include)
  src/apps/CMakeLists.txt:17 (find_package)

```

I tried using the provided build [https://drive.google.com/drive/folders/1Ah580EIylJJ0v2vGOeSBU_b8zPDWESxS](url), but it didn't work either.

jfelixre, 2025-01-28:
I have the same problem

sickopickle, 2025-01-29:
I reinstalled Ubuntu and followed the steps exactly and it still didn't work.

sickopickle, 2025-02-04:
@jfelixre 
I used this link and followed instructions from brad, and it worked, although I am currently still struggling with low FPS
https://github.com/Genesis-Embodied-AI/Genesis/issues/207

closed 2025-02-04 by sickopickle