Genesis-Embodied-AI/genesis-world #589: Support capsule and better error messages?

dementrock, 2025-01-16:
Is there any plan to support capsule as one of the primitive geoms? So that there's better feature parity with Mujoco.

Also right now, loading URDF with capsule geom gives a cryptic error message like below:

```
  File "/opt/conda/envs/ros_env/lib/python3.11/site-packages/genesis/ext/urdfpy/urdf.py", line 144, in _parse_simple_elements
    v = [t._from_xml(n, path) for n in vs]
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/ros_env/lib/python3.11/site-packages/genesis/ext/urdfpy/urdf.py", line 144, in <listcomp>
    v = [t._from_xml(n, path) for n in vs]
         ^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/ros_env/lib/python3.11/site-packages/genesis/ext/urdfpy/urdf.py", line 1001, in _from_xml
    kwargs = cls._parse(node, path)
             ^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/ros_env/lib/python3.11/site-packages/genesis/ext/urdfpy/urdf.py", line 168, in _parse
    kwargs.update(cls._parse_simple_elements(node, path))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/ros_env/lib/python3.11/site-packages/genesis/ext/urdfpy/urdf.py", line 136, in _parse_simple_elements
    v = t._from_xml(v, path)
        ^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/ros_env/lib/python3.11/site-packages/genesis/ext/urdfpy/urdf.py", line 188, in _from_xml
    return cls(**cls._parse(node, path))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/ros_env/lib/python3.11/site-packages/genesis/ext/urdfpy/urdf.py", line 662, in __init__
    raise ValueError("At least one geometry element must be set")
```

Would be better if it more explicitly says capsules are not supported yet.

zhouxian, 2025-01-17:
is capsule a standard element in URDF? In my understanding it's actually not
let me know if there's any reference

dementrock, 2025-01-17:
@zhouxian ah good point. Indeed it was not mentioned in https://wiki.ros.org/urdf/XML/link. However, we have seen it in some of the robot definitions (perhaps the vendor just used urdf as format but imported it to mujoco?). A bit tricky for me to find public examples, but one was referenced in https://github.com/bulletphysics/bullet3/blob/e9c461b0ace140d5c73972760781d94b7b5eee53/data/capsule.urdf#L16

Genesis-Embodied-AI, 2025-01-18:
ok i see
we used urdfpy for parsing urdfs, i think we should be able to add capsule

dementrock, 2025-01-19:
Thanks @Genesis-Embodied-AI ! Also thanks for reminding me that actually the mujoco import supports capsule. I am unblocked right now so this is not urgent for me, but seems like good to support if easy so there's more parity with mujoco.

duburcqa, 2025-05-24:
Related to https://github.com/Genesis-Embodied-AI/Genesis/issues/564