Genesis-Embodied-AI/genesis-world #592: [Feature] Introduce templates to encourage useful issues and pull requests

lgleim, 2025-01-17:
While the [CONTRIBUTING](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/CONTRIBUTING.md) guidelines already provide significant guidance on how to structure Bug reports & Feature requests, the large majority of current issue reports does not follow them. 

To ensure more structured & time efficient project management, I see quite some benefit in [using templates to encourage useful issues and pull requests](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests).

This would make it easier for new users to contribute in a structured manner by guiding them in creating useful reports in alignment with the contribution guide, thus

- saving reviewers time
- improving the quality of issue reports and feature requests
- direct questions to the community discussion page
- ensure required PR preparation steps were honored

Kashu7100, 2025-01-17:
Yeah, I had the same concern for a while. Let me work on this.

lgleim, 2025-01-17:
I prepared one for you to review yesterday in my fork. will create PR asap

closed 2025-01-22 by Kashu7100