Genesis-Embodied-AI/genesis-world #599: splashsurf error

ZhihongLiao, 2025-01-17:
I already install splashsurf, but it still does not work.
Traceback (most recent call last):
  File "/home/lzh/Genesis-main/examples/tutorials/sph_liquid.py", line 49, in <module>
    scene.step()
  File "/home/lzh/Genesis-main/genesis/utils/misc.py", line 48, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lzh/Genesis-main/genesis/engine/scene.py", line 727, in step
    self._visualizer.update(force=False)
  File "/home/lzh/Genesis-main/genesis/vis/visualizer.py", line 135, in update
    self._viewer.update()
  File "/home/lzh/Genesis-main/genesis/vis/viewer.py", line 113, in update
    buffer_updates = self.context.update()
                     ^^^^^^^^^^^^^^^^^^^^^
  File "/home/lzh/Genesis-main/genesis/vis/rasterizer_context.py", line 721, in update
    self.update_sph(buffer_updates)
  File "/home/lzh/Genesis-main/genesis/vis/rasterizer_context.py", line 465, in update_sph
    mesh = pu.particles_to_mesh(
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/lzh/Genesis-main/genesis/utils/particle.py", line 311, in particles_to_mesh
    gs.raise_exception(f"Failed to import splashsurf. {splashsurf_error_msg}")

[Genesis] [00:02:26] [ERROR] Failed to import splashsurf. FileNotFoundError: [Errno 2] No such file or directory: 'splashsurf'
[Genesis] [00:02:26] [INFO]  Exiting Genesis and caching compiled kernels...

futuyao, 2025-01-22:
Same error here. Any solutions?

futuyao, 2025-01-22:
Nvm. Just need to install splashsurf as https://github.com/InteractiveComputerGraphics/splashsurf

duburcqa, 2025-05-24:
Error handling has been improved. What to do in this case should be more clear now.

closed 2025-05-24 by duburcqa