Genesis-Embodied-AI/genesis-world #600: Particles Passing Through Rigid Body in MPM Simulation

hooked-on-mas, 2025-01-17:
Hello,
Thank you for creating and sharing such an amazing tool.

Currently, I am working on a simulation involving the interaction between granular particles and rigid bodies. 
However, I encountered an issue where **particles seem to pass through the rigid body,** as shown in the video linked below.

While the simulation works well with SPH, I couldn't achieve the desired behavior with MPM—the particles fail to stay within the box.
Is this a limitation of MPM itself, or might there be something I am overlooking?

If you could point out areas for improvement, I’d be happy to understand and contribute.
Thank you for your guidance and support!

https://github.com/user-attachments/assets/cc58cb5c-88ce-428d-9127-876cbf16ad5b

https://github.com/user-attachments/assets/75a6df08-d3da-4d0d-a27b-dab983a39894

hooked-on-mas, 2025-01-17:
```Python
import argparse

import numpy as np

import genesis as gs

def main():
    ########################## init ##########################
    gs.init(seed=0, precision="32", logging_level="debug")

    ########################## create a scene ##########################
    scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            dt=3e-3,
            substeps=10,
        ),
        mpm_options=gs.options.MPMOptions(
            lower_bound=(0.0, -1.0, -0.1),
            upper_bound=(0.57, 1.0, 2.4),
            grid_density=64,
        ),
        sph_options=gs.options.SPHOptions(
            lower_bound=(0.0, -1.0, -0.1),
            upper_bound=(0.57, 1.0, 2.4),
        ),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(4.5, 0.0, 1.42),
            camera_lookat=(1.0, 0.0, 1.0),
            camera_fov=30,
            max_FPS=120,
        ),
        vis_options=gs.options.VisOptions(
            visualize_mpm_boundary=True,
            visualize_sph_boundary=True,
        ),
    )

    plane = scene.add_entity(
        material=gs.materials.Rigid(
            needs_coup=True,
            coup_friction=0.2,
        ),
        morph=gs.morphs.URDF(file="urdf/plane/plane.urdf", fixed=True),
    )
    mat_wheel = gs.materials.Rigid(
        needs_coup=True,
        coup_softness=0.0,
    )
    wheel_0 = scene.add_entity(
        material=mat_wheel,
        morph=gs.morphs.URDF(
            file="../models/cup/cup_wheel.urdf",
            pos=(0.5, 0.0, 1.6),
            euler=(0, 0, 90),
            scale=0.2,
            convexify=False,
            fixed=True,
        ),
    )

    emitter = scene.add_emitter(
        material=gs.materials.SPH.Liquid(mu=0.01),
        # material=gs.materials.MPM.Sand(),
        max_particles=200000,
        surface=gs.surfaces.Rough(
            color=(1.0, 0.9, 0.6, 1.0),
        ),
    )
    scene.build()

    horizon = 1000
    for i in range(horizon):
        print(i)
        emitter.emit(
            pos=np.array([0.5, 0.0, 2.3]),
            direction=np.array([0.0, 0.0, -1.0]),
            speed=8.0,
            droplet_shape="rectangle",
            droplet_size=[0.03, 0.05],
        )
        scene.step()

if __name__ == "__main__":
    main()

```

wangyian-me, 2025-01-18:
The different between mpm-rigid and sph-rigid coupling is that, the mpm-rigid use grid in mpm to collide with rigid while the sph directly use particles. Thus, it's better to make sure that your grid is smaller than you rigid object. (your boundary might not be thicker than (2.4 + 0.1) / 64 = 0.04).

One possible solution is (not 100% sure): set `grid_density` larger (to 128 or 256 for example). And set the rigid object with larger scale.

hooked-on-mas, 2025-01-20:
Thank you for your suggestion! As you mentioned, I set the grid_density to 128, and it almost resolved the issue of objects passing through. I really appreciate your help!

https://github.com/user-attachments/assets/5654e96e-3292-422f-96c0-7cb2882e372c

closed 2025-01-20 by hooked-on-mas