Genesis-Embodied-AI/genesis-world #602: weird collision behavior when simulating an object fall down

HilbertXu, 2025-01-17:
Hi, thanks for this amazing work! 

I'm currently trying to build a robot manipulation env in Genesis with object models from YCB dataset. It works fine if I only let those objects falls on the ground plane (`gs.morphs.Plane()`), but it totally failed when I add an table plane through (`gs.morphs.Box(size = np.asarray([1.0, 1.0, 0.0001]), pos = np.asarray([0.5, 0.5, 0.15]), fixed=True)`). The added objects fall down to the box plane and instantly fly away. I noticed that this behavior happens to non-convex object model like scissors. if you play the attached video on x0.25 speed, you can see that the scissors suddenly fly away without any force applied :(

I haven checked the collision model of each loaded object, all of them look fine. The scissor model is convexified nicely. Please see the attached images. 

[image]

[image]

I was kinda suspecting that this behavior might be related to the little shaking of the loaded objects as mentioned in #444 
thus I tried the solution in it: turning on the decompose_nonconvex and set the coacd_options to `gs.options.CoacdOptions(threshold=0.01, preprocess_resolution=100)`, but this doesn't help. 

https://github.com/user-attachments/assets/0ece1ae7-9547-44d9-bcd9-ee37ae5c4b47

Could you please kindly help me to figure out what is the actual reason of this behavior and how to fix it? If any further information is needed, please let me know :)

Cheers,
Yucheng

HilbertXu, 2025-01-17:
I recorded another video with collision view of the box table plane and objects, hope it can provide more information. 

https://github.com/user-attachments/assets/c50f2e2b-49e2-4101-928c-303bc12aa6e9

Kashu7100, 2025-01-18:
@HilbertXu Could you provide your sample objects (maybe in .zip) with corresponding script to reproduce the error?

@YilingQiao could you take a look?

HilbertXu, 2025-01-18:
Hi @Kashu7100 
here is the sample objects, I think I probably found out why this issue happens. 

[debug.zip](https://github.com/user-attachments/files/18464583/debug.zip)

I tried with the `textured.obj` not the `textured_simple.obj` one I used before, `textured_simple.obj` is a simplified version comes from YCB dataset. It behaves differently, the object will directly fall and penetrate the plane. Then I start to suspecting that this problem comes from two aspects: (1) the simplified model from YCB dataset is not good enough, especailly for non-convex models; (2) The box plane I used in my env is too thin. 

Then, I recalled that I met the same "penetration" issue with Blender before, and I made the plane thicker to solve that problem. So I did the same thing with the box plane in Genesis, oops, problem solved. The Mesh loaded from `textured.obj` fall down to the box plane with size (1.0, 1.0, 0.1). But the object is still shaking on the table plane...

https://github.com/user-attachments/assets/ec2d6618-1923-4ecf-ba21-d35dd25701a9

Although my initial issue can be addressed with thicker plane and an unsimplified mesh model, I'm still curious why this seems to be a common case in different physical simulators, maybe you guys have some ideas? Also, as I have to make the table plane thicker, I have to take its thickness into consideration. I guess I can directly call the 'entity.get_pos()' function? Does it return the center of mass of an entity? What if that entity is not a primitive shape, for example, a table plane with four cylinder feet? 

Cheers,
Yucheng

Kashu7100, 2025-01-18:
This may help https://github.com/Genesis-Embodied-AI/Genesis/issues/444

wangyian-me, 2025-01-18:
Collision with Thin-Shell is hard unless specially treated, since the objects goes to the opposite side once penetrated rather than having a deeper penetration.

wangyian-me, 2025-01-18:
To prevent shaking, a simple method is to set the damping of the dofs to a larger value (maybe 0.5).

duburcqa, 2025-05-24:
@HilbertXu Could you try again on main branch? I think this issue has been fixed now.

closed 2025-05-24 by duburcqa