Genesis-Embodied-AI/genesis-world #603: Mujoco import requires conaffinity and contype

danx0r, 2025-01-17:
importing a mujoco (mjcf) model like so:
```
gs.morphs.MJCF(file='mujocomodel.xml')
```
Expected result: all elements of the Mujoco model imported

Actual result: geoms are ignored (do not load) unless I explicitly define the attributes ```conaffinity="0" contype="0"```

Kashu7100, 2025-01-18:
Please refer to https://github.com/Genesis-Embodied-AI/Genesis/issues/287#issuecomment-2562311658 for the current supported objects in mujoco. 

@danx0r Can you provide an example .xml (with object file if applicable) that you meet the problem?

danx0r, 2025-01-18:
There is no need to upload an example, this behavior is exactly what is discussed in #287. It's a matter of deciding what you will support, but also what defaults to use. It is counter-intuitive to import a model and have it not visualize all the parts. At a minimum there should be some sort of warning message.

Possibly if I had RTFM closely I would have seen this caveat, but honestly I don't see why this would be the default behavior. It is in my experience standard practice to have mesh objects visualize even if they don't collide, so I would suggest that contype=conaffinity=0 be the default, and then ppl will seek out docs to explain how to add collision meshes or primitives.

closed 2025-03-15 by zswang666