Genesis-Embodied-AI/genesis-world #609: OpenGL.error.Error: Attempt to retrieve context when no valid context

siddarth09, 2025-01-18:
`Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/siddarth/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1161, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/siddarth/.local/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/home/siddarth/.local/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/siddarth/.local/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/siddarth/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 955, in _time_event
    self.on_draw()
  File "/home/siddarth/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._render()
  File "/home/siddarth/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1098, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/home/siddarth/.local/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/siddarth/.local/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/siddarth/.local/lib/python3.10/site-packages/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "src/latebind.pyx", line 51, in OpenGL_accelerate.latebind.Curry.__call__
  File "/home/siddarth/.local/lib/python3.10/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 469, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/siddarth/.local/lib/python3.10/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
  File "/home/siddarth/.local/lib/python3.10/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context

`

am running the hello genesis script given below 

import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()

All I get is a blank screen with the above error. Any help would be appreciated

OKsusu1411, 2025-01-21:
I had exact same error you've had. I solved this error by define environment variables at terminal which want to run genesis    
```shell
export MUJOCO_GL=glx
export PYOPENGL_PLATFORM=glx
```
after export then try your python code.   
if you bothering to type those command every time you open new terminal. you can just edit .bashrc yourself   
hope it helps you a bit

siddarth09, 2025-01-21:
Thanks a lot, I will be closing this issue