Genesis-Embodied-AI/genesis-world #613: Simulation of forces and collision on Genesis?

MilanaAmighiVUB, 2025-01-20:
Hello,

I'm interested in the forces and collision topic, but it seems to no be released yet (it says coming soon).

Do you know more or less when this option will be available?

zswang666, 2025-01-22:
They are both supported. The coming soon thingy is referring to the documentation not the code.

MilanaAmighiVUB, 2025-01-22:
Hello,

Thank you for the answer. Do you know where I can find some documentation about that? Also, do you know if the force and collision simulation works with soft robots?

zswang666, 2025-02-13:
For (contact, I supposed) forces and collisions, you can find the basic API [here](https://github.com/Genesis-Embodied-AI/Genesis/blob/e7f98d9360e6ab0d60f6cffcb902ae583f31b72c/genesis/engine/entities/rigid_entity/rigid_entity.py#L2256). Also, briefly, the relevant files include
- algorithms for collision detections: [collider_decomp.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/engine/solvers/rigid/collider_decomp.py)
- geometry representation for collision check: [mpr_decomp.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/engine/solvers/rigid/mpr_decomp.py) and [sdf_decomp.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/engine/solvers/rigid/sdf_decomp.py)
- contact dynamics: [constraint_solver_decomp.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/engine/solvers/rigid/constraint_solver_decomp.py)

Also regarding force and collision for soft robots,
- MPM naturally supports self-collision; also coupling with rigid is also supported
- FEM coupling with rigid is supported but no self-collision yet

pushchar, 2025-02-14:
Is there an estimated timeline for the documentation release?
Thanks!

benediktstroebl, 2025-05-06:
Bumping this. Is there a timeline for when the documentation on this might be released? Thank you!