Genesis-Embodied-AI/genesis-world #614: [Feature Request] Enable CoACD in URDF for the links of a robotic arm

alexis779, 2025-01-20:
Hi,

I have been following genesis tutorial to control a robotic arm to pickup a rigid ball.

I'm running into penetration issue when the jaw is holding the ball. See Video 4 in [Google Photo album](https://photos.app.goo.gl/spjm9iKpBp5gMLKC9). The robotic arm (SO-ARM-100) is loaded from URDF configuration. You can find the mesh for the mobile jaw for example in [this file](https://github.com/TheRobotStudio/SO-ARM100/blob/main/URDF/SO_5DOF_ARM100_8j_URDF.SLDASM/meshes/Moving%20Jaw.STL).

Currently the collision mesh is accurately displaying the visual shape of the jaw links, after I disabled `convexify` option. Compare Video 5 against Video 4.

But the jaw is squishing the rigid ball like a marshmallow.

I have been reading through this [issue](https://github.com/Genesis-Embodied-AI/Genesis/issues/444) and related threads describing a floating object laid on a table, while it should remain still. The current issue is related but slightly different as it affects a solid object (sphere against jaw links) and not a thin-shell (scissors over a table). I have also experimented with increasing damping, which mostly slows down the motion of the mobile jaw.

Here the solution is the same as the one recommended in this [message](https://github.com/Genesis-Embodied-AI/Genesis/issues/444#issuecomment-2581136017). However I don't think it's available right now with the current API.

@wangyian-me can `decompose_nonconvex` / `coacd_options` parameters be made available for `URDF`? It's only available for `Morph`. This would apply to all the links in the robotic arm, if that makes sense / is feasible?

I only see `convexify` parameter available in URDF [list](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/options/morphs.py#L508). Does this mean SDF is the only possible option for collision detection  ?

I'd be curious to test out URDF with CoACD enabled.

MandiZhao, 2025-01-21:
my current solution is writing a script that manually parses & rewrites your urdf/xml:
for each object mesh, run CoACD to get a list of convex mesh files, and add each convex mesh part as a new geom/link to the new urdf, then load the new urdf. [obj2mjcf](https://github.com/kevinzakka/obj2mjcf) is basically does this for xmls.

alexis779, 2025-01-21:
Thanks @MandiZhao, For now I'll switch from URDF to the script-generated MJCF config. According to the 4. item in [SO-ARM-100 MJCF config](https://github.com/google-deepmind/mujoco_menagerie/tree/main/trs_so_arm100), the convex decomposition should have already been done for SO-ARM-100.

alexis779, 2025-01-24:
With the approximation of the [fixed jaw](https://github.com/google-deepmind/mujoco_menagerie/blob/main/trs_so_arm100/assets/Fixed_Jaw.stl) into [a small set of convex shapes](https://github.com/google-deepmind/mujoco_menagerie/blob/main/trs_so_arm100/so_arm100.xml#L106-L111), the results are much better. The ball is stabilized during grasping.

See video 6 in the [photo album](https://photos.app.goo.gl/spjm9iKpBp5gMLKC9) for the demo with the MJCF config.

Here's an illustration of the collision mesh formed by 2 main piece (green, blue) + 4 collision pads (cyan, magenta, yellow, red)

[image]

Here's the actual link.
[image]

duburcqa, 2025-03-28:
On it.

closed 2025-03-29 by duburcqa