Genesis-Embodied-AI/genesis-world #615: How to interact with a scene such as adding box?

robot-gen, 2025-01-20:
Hi,

In my case, I would like to add box after a scene built, such as keep adding boxes as obstacles in the scene. How to achieve this? 

What I have tried is to call scene.add_entity(gs.morphs.Box()), while getting error saying " gs.raise_exception("Scene is already built.")".

Thanks!

bxtbold, 2025-01-22:
As far as I know, it might not be available because Genesis compiles kernels every time you build a scene, which involves memory allocation etc.

It is stated in [here](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html#build-the-scene-and-start-simulating)

YilingQiao, 2025-02-02:
Thanks @bxtbold ! Yeah, adding objects during simulation is not supported. However, there might be a workaround. Can you initialize all the objects you need at the beginning and place them very far away? Then, move them into the scene when needed.

closed 2025-02-02 by YilingQiao