Genesis-Embodied-AI/genesis-world #616: Bug. Error occurs when load MJCF.

Riften, 2025-01-20:
**Description**
`ValueError: zero-size array to reduction operation maximum which has no identity` when load the mjcf model attached.

[floor_backing_room.zip](https://github.com/user-attachments/files/18477338/floor_backing_room.zip)

<details>
<summary>Full output</summary>

```bash
[Genesis] [20:39:22] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [20:39:22] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [20:39:22] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [20:39:22] [INFO] Running on [NVIDIA GeForce GTX 1080 Ti] with backend gs.cuda. Device memory: 10.91 GB.
[Genesis] [20:39:22] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [20:39:23] [INFO] Scene <7385b01> created.
[Genesis] [20:39:23] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <b2ee330>, morph: <gs.morphs.MJCF(file='/home/yongxi/Workspace/SJTU-Robotic-Interface/notebooks/floor_backing_room/model.xml')>, material: <gs.materials.Rigid>.
/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/numpy/core/fromnumeric.py:3504: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/numpy/core/_methods.py:121: RuntimeWarning: invalid value encountered in divide
  ret = um.true_divide(
Traceback (most recent call last):
  File "/home/yongxi/Workspace/SJTU-Robotic-Interface/notebooks/reproduce.py", line 4, in <module>
    scene.add_entity(gs.morphs.MJCF(file='floor_backing_room/model.xml'))
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/scene.py", line 359, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/simulator.py", line 134, in _add_entity
    entity = self.rigid_solver.add_entity(self.n_entities, material, morph, surface, visualize_contact)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/solvers/rigid/rigid_solver_decomp.py", line 64, in add_entity
    entity = entity_class(
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 68, in __init__
    self._load_model()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 78, in _load_model
    self._load_MJCF(self._morph, self._surface)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 342, in _load_MJCF
    g_info = mju.parse_geom(mj, i_g, morph.scale, morph.convexify, surface, morph.file)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/utils/mjcf.py", line 357, in parse_geom
    mesh = gs.Mesh.from_trimesh(
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/mesh.py", line 260, in from_trimesh
    opacity_texture = color_texture.check_dim(3)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/options/textures.py", line 149, in check_dim
    return ImageTexture(image_array=res, image_color=res_color, encoding="linear").check_simplify()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/options/textures.py", line 155, in check_simplify
    max_color = np.max(self.image_array, axis=(0, 1))
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/numpy/core/fromnumeric.py", line 2810, in max
    return _wrapreduction(a, np.maximum, 'max', axis, None, out,
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/numpy/core/fromnumeric.py", line 88, in _wrapreduction
    return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
ValueError: zero-size array to reduction operation maximum which has no identity
[Genesis] [20:39:23] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

</details>

**To Reproduce**
```python
import genesis as gs
gs.init(backend=gs.cuda)
scene = gs.Scene(show_FPS=False, show_viewer=False)
scene.add_entity(gs.morphs.MJCF(file='floor_backing_room/model.xml'))
```

**Expected behavior**
Result from mujoco 3.2.5.

[image]

**Environment:**
 - OS: ubuntu 20.04
 - GPU/CPU: NVIDIA GeForce GTX 1080 Ti

bxtbold, 2025-01-22:
I was able to reproduce and find out where the bug was appearing.

Basically, the the image was loaded properly at [line 305 of utils/mjcf.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/36b6670a4a0783d4ed58d6cd8edbd5662727677f/genesis/utils/mjcf.py#L305), but somehow the image_array became an empty array at [line 307](https://github.com/Genesis-Embodied-AI/Genesis/blob/36b6670a4a0783d4ed58d6cd8edbd5662727677f/genesis/utils/mjcf.py#L307). This is the reason why you having that value error (numpy reduction functions cannot process empty array).

If you comment out line 307, you can load that mjcf model.

[image]

Riften, 2025-01-22:
@bxtbold Thank you! It seems that the bug is caused by incorrect processing of the `texrepeat` attribute in MJCF. In the MJCF file there is `<material texrepeat="0.5 0.5" texuniform="true" name="floor_backing_room_wall_mat" texture="floor_backing_room_wall" />`. However, `mjcf.py` converts texrepeat to `int` which results in zero repeat.

```python
tex_repeat = mj.mat_texrepeat[mat_id].astype(int) # [0, 0]
image_array = np.tile(image_array, (tex_repeat[0], tex_repeat[1], 1))
```

closed 2025-02-05 by YilingQiao