Genesis-Embodied-AI/genesis-world #619: mjcf doesn't parse actuator ctrlrange properly

MandiZhao, 2025-01-21:
When loading the h1 model from [DIAL-MPC](https://github.com/LeCAR-Lab/dial-mpc) the actuator force range does not get properly parsed and throws warnings ```[WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.```  I attached the h1 model assets and xml here, run `load_h1.py` to reproduce: 
[h1_mjcf_loading.zip](https://github.com/user-attachments/files/18485358/h1_mjcf_loading.zip)

Full error msg below:

```
[Genesis] [21:41:19] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [21:41:19] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [21:41:19] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [21:41:19] [INFO] Running on [NVIDIA GeForce RTX 4080 SUPER] with backend gs.cuda. Device memory: 15.70 GB.
[Genesis] [21:41:19] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [21:41:19] [INFO] Scene <8e5f0b1> created.
[Genesis] [21:41:19] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <774fe82>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [21:41:19] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <2a5693d>, morph: <gs.morphs.MJCF(file='/home/mandiz/pr-genesis/unitree_h1/h1_real_feet.xml')>, material: <gs.materials.Rigid>.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
[Genesis] [21:41:19] [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.
```

Hippozhibos, 2025-01-22:
get same WARNING when using Genesis

bxtbold, 2025-01-23:
> When loading the h1 model from [DIAL-MPC](https://github.com/LeCAR-Lab/dial-mpc) the actuator force range does not get properly

First of all, xml model doesn't provide force range. Instead, it provides control range.

> [WARNING] `kp` in `gainprm` doesn't match `-kp` in `biasprm`.

This warning has nothing to do with force or control range, but it is about control gains.
Both gainprm kp and biasprm kp gains are same for [actuator/position](https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator-position). Not really sure why those values enforced to be equal.

If you wanna add force range, you need to add force range argument to your actuators

```diff
-    <motor class="h1" name="left_hip_yaw" joint="left_hip_yaw" ctrlrange="-200 200"/>
+    <motor class="h1" name="left_hip_yaw" joint="left_hip_yaw" forcerange="-200 200" ctrlrange="-200 200"/>
```

duburcqa, 2025-03-22:
Genesis only supports PD control for now, hence this requirement. This may change in the future.