Genesis-Embodied-AI/genesis-world #624: There is an issue with the simulation collision.

Xu-Xuanhui, 2025-01-22:
[image]
I defined the visual and collision in the URDF file using Blender. The collision uses the mesh, but during simulation in Genesis, the collision does not closely adhere to the visual surface and instead forms an outer bounding triangle. In Isaacsim, I can resolve this by changing the Collider's approximation format from TriangleMesh to SDFMesh. I would like to know how to address this problem in Genesis?

zswang666, 2025-01-22:
This is probably because by default, meshes are converted to convex hulls in Genesis for efficiency. Could you try to turn that off by
```
entity = scene.add_entity(
    morph=gs.morphs.Mesh(
        ...
        convexify=False,
    ),
)
```
This will trigger convex decomposition instead.

wangyian-me, 2025-01-25:
It's a URDF file, so use `gs.morphs.URDF `and also set `convexify=False` in options.

zswang666, 2025-02-13:
Is this issue solved?
Also potentially related: https://github.com/Genesis-Embodied-AI/Genesis/issues/614

duburcqa, 2025-03-23:
Closing due to inactivity.

closed 2025-03-23 by duburcqa