Genesis-Embodied-AI/genesis-world #625: [Bug]: Unitree Go2 gets “launched away” when spawned on complex terrain generated by the terrain generator

Adwaver4157, 2025-01-22:
### Bug Description

Hello, I’m currently encountering an issue with the Unitree Go2 robot model when using a terrain generator(https://github.com/leggedrobotics/terrain-generator) to create complex terrains. As soon as I spawn the Go2 on this generated mesh, the robot instantly flies off to some distant location. Calling the reset function afterwards does not resolve the problem—once it’s launched away, the robot cannot be properly reset.

Context / Steps to Reproduce:

I have a trained model (using go2_eval.py) that was originally trained on a flat plane.
In go2_env.py, I load a mesh generated by the terrain generator to create a more complex terrain.
Immediately after launching the simulator in go2_eval.py, the robot gets flung away, and calling reset() does not fix the situation.
Additional Details:

The training on a flat plane works without issues.
The problem only arises when using the newly generated terrain mesh.
Once the robot is “launched” upon spawn, it’s impossible to reset it properly.
Any guidance or suggestions on how to prevent this unexpected behavior would be greatly appreciated. Thank you!

[Launched Unitree Go2](https://drive.google.com/file/d/1zVtNEZcH-OG30PuajSKPUHY4hXA61qHz/view?usp=sharing)
[Mesh I used](https://drive.google.com/file/d/1vHEbWtCdI7oYplFvtRH2S5yP0vQgPus7/view?usp=sharing)

### Steps to Reproduce

```python
# I only modified go2_env.py to add complex terrain mesh.
import torch
import math
import genesis as gs
from genesis.utils.geom import quat_to_xyz, transform_by_quat, inv_quat, transform_quat_by_quat

def gs_rand_float(lower, upper, shape, device):
    return (upper - lower) * torch.rand(size=shape, device=device) + lower

class Go2Env:
    def __init__(self, num_envs, env_cfg, obs_cfg, reward_cfg, command_cfg, show_viewer=False, device="cuda"):
        self.device = torch.device(device)

        self.num_envs = num_envs
        self.num_obs = obs_cfg["num_obs"]
        self.num_privileged_obs = None
        self.num_actions = env_cfg["num_actions"]
        self.num_commands = command_cfg["num_commands"]

        self.simulate_action_latency = True  # there is a 1 step latency on real robot
        self.dt = 0.02  # control frequency on real robot is 50hz
        self.max_episode_length = math.ceil(env_cfg["episode_length_s"] / self.dt)

        self.env_cfg = env_cfg
        self.obs_cfg = obs_cfg
        self.reward_cfg = reward_cfg
        self.command_cfg = command_cfg

        self.obs_scales = obs_cfg["obs_scales"]
        self.reward_scales = reward_cfg["reward_scales"]

        # create scene
        self.scene = gs.Scene(
            sim_options=gs.options.SimOptions(dt=self.dt, substeps=2),
            viewer_options=gs.options.ViewerOptions(
                max_FPS=int(0.5 / self.dt),
                camera_pos=(2.0, 0.0, 2.5),
                camera_lookat=(0.0, 0.0, 0.5),
                camera_fov=40,
            ),
            vis_options=gs.options.VisOptions(n_rendered_envs=1),
            rigid_options=gs.options.RigidOptions(
                dt=self.dt,
                constraint_solver=gs.constraint_solver.Newton,
                enable_collision=True,
                enable_joint_limit=True,
            ),
            show_viewer=show_viewer,
[... 211 lines cut ...]
        return torch.square(self.base_pos[:, 2] - self.reward_cfg["base_height_target"])

```

### Expected Behavior

Go2 starts walking on complex terrain without being launched (though it may stumble)

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 20.04
- GPU/CPU: RTX3090
- GPU-driver version: 550
- CUDA / CUDA-toolkit version: 12.1

### Release version or Commit ID

Genesis version: 0.2.1
Commit ID: 1f37930cef144b6220a738448330dbeb6fa9f2fa

Kashu7100, 2025-01-22:
Could you check if your initial robot location is a "free space", meaning no collision is happening?

Adwaver4157, 2025-01-23:
Even when starting the robot from a plain surface instead of on the mesh, it suddenly stops near the mesh. Based on the normals, there should not be any obstacles.

https://github.com/user-attachments/assets/03c53d46-ae75-4bd7-8ffe-fe026d6ee279
https://github.com/user-attachments/assets/0bb63b27-fdbd-4519-b058-9634887259f2

Kashu7100, 2025-01-24:
I think this is something to do with convexify option. Can you try to set `convexify=False` for your asset?

```python
morph=gs.morphs.Mesh(
        ...
        convexify=False,
    ),
```

Nate711, 2025-01-25:
I have the same problem dropping my quadruped robot onto complex terrain. Inevitably with higher number of envs (10k) at least one robot will give all NaN for state after colliding with the ground. What happens is the robot will collide with the ground and then shoot off into infinity. Happens with both Newton and CG. As you can imagine, Genesis is unusable with this problem since the whole simulation crashes.

[image]

Nate711, 2025-01-25:
@Kashu7100 When using the terrain generator `morphs.Terrain` there doesn't seem to be a convexify option since you don't directly create it as a mesh?

Kashu7100, 2025-02-03:
Can you check your `dt` settings? (As pointed in #653 )

Nate711, 2025-02-17:
@Kashu7100 Thanks for the response! I tried decreasing dt to as low as 0.0004s but I still see robots fly away or have velocities that become abnormally high. 

Please try this script to reproduce the issue: https://github.com/Nate711/Genesis-backflip/blob/main/flying_away_bug.py 
In this script I reset the position, orientation and dofs of any robots whose link velocities exceed 100m/s. However as you can see from the video below, many robots still fly off into space!

https://github.com/user-attachments/assets/34941475-88db-4bab-8d12-b54bde9ebcd1

lupinjia, 2025-02-23:
I have the similar issue, though I initialize the robot at a relatively lower height. But the environment just returns nan values in base_pos, dof_pos, dof_vel and other robot states.
I have provided the code to reproduce in https://github.com/lupinjia/genesis_lr/tree/self-imple. 
# My test results
- For num_envs=10000, use_implicit_controller=False, dt=0.02, decimation=4
  
  The training session terminates at ite 24. I implemented a rendered_envs_idx utility as https://github.com/Genesis-Embodied-AI/Genesis/pull/752 and rendered the abnormal env whose index is 9405. The result is as the following video, it just scratchs on the ground, trying to learn to walk, and suddenly the nan values come.

https://github.com/user-attachments/assets/616c1e99-cff9-4d29-80f3-4cdbe927ef45

- For num_envs=8192, use_implicit_controller=False, dt=0.02, decimation=4

  The nan iteration becomes 1532
- For num_envs=10000, use_implicit_controller=False, dt=0.005, decimation=4
  
  The nan iteration becomes 726

[image]

- For num_envs=10000, use_implicit_controller=True, dt=0.02, decimation=4

  The nan iteration becomes 749

[image]

# Comments
Though the policy can learn to walk before nan values come out under some configuration, but it's unfavorable for those who want to continue training to get better robustness and performance if nan values come out suddenly.

closed 2025-09-27 by duburcqa