Genesis-Embodied-AI/genesis-world #633: [Bug]:

irrikrlla, 2025-01-25:
### Bug Description

Hi, I am new here. I am trying to run the code in Chapter 'Hello Genesis' to generate  a Franka arm but I got this unexpected error when running sence.build, the following is a full log:

>>> import genesis as gs
>>> gs.init(backend=gs.gpu)
[Genesis] [09:36:44] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [09:36:44] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [09:36:44] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [09:36:44] [INFO] Running on [NVIDIA GeForce RTX 4070 Ti SUPER] with backend gs.cuda. Device memory: 15.99 GB.
[Genesis] [09:36:44] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
>>> scene = gs.Scene(show_viewer=True)
[Genesis] [09:36:53] [INFO] Scene <3f11d38> created.
>>> plane = scene.add_entity(gs.morphs.Plane())
[Genesis] [09:37:02] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <79c5c43>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [09:37:02] [INFO] Preprocessing geom idx 0.
>>>franka = scene.add_entity(
...     gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
... )
[Genesis] [09:37:17] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <7da1557>, morph: <gs.morphs.MJCF(file='/home/libra/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [09:37:17] [INFO] Preprocessing geom idx 1.
[Genesis] [09:37:18] [INFO] Preprocessing geom idx 2.
[Genesis] [09:37:18] [INFO] Preprocessing geom idx 3.
[Genesis] [09:37:18] [INFO] Preprocessing geom idx 4.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 5.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 6.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 7.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 8.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 9.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 10.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 11.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 12.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 13.
[Genesis] [09:37:19] [INFO] Preprocessing geom idx 14.
[Genesis] [09:37:20] [INFO] Preprocessing geom idx 16.
>>> scene.build()
[Genesis] [09:37:32] [INFO] Building scene <3f11d38>...
[Genesis] [09:37:39] [INFO] Compiling simulation kernels...
[Genesis] [09:38:00] [INFO] Building visualizer...
Dropped Escape call with ulEscapeCode : 0x03007703
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/libra/Genesis/genesis/ext/pyrender/viewer.py", line 1160, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/libra/.local/lib/python3.11/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/libra/.local/lib/python3.11/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/libra/.local/lib/python3.11/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/libra/Genesis/genesis/ext/pyrender/viewer.py", line 954, in _time_event
    self.on_draw()
  File "/home/libra/Genesis/genesis/ext/pyrender/viewer.py", line 653, in on_draw
    self._render()
  File "/home/libra/Genesis/genesis/ext/pyrender/viewer.py", line 1097, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/libra/Genesis/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/libra/Genesis/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/libra/Genesis/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/libra/.local/lib/python3.11/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/libra/.local/lib/python3.11/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 465, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/libra/.local/lib/python3.11/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
              ^^^^^^^^^^^^^^^^^^^^^
  File "/home/libra/.local/lib/python3.11/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init(backend=gs.gpu)
scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
...     gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
... )
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

build and display the franka arm sence porperly

### Relevant log output

```shell

```

### Environment

- OS:  Windows 11 24H2,  WSL Ubuntu 22.04( because windows is poor supported ) 
- GPU/CPU [RTX 4070 Ti Super/Ryzen 7900x] 
- CUDA / CUDA-toolkit version : 12.4

### Release version or Commit ID

a): version : latest;
b): git head :806d0a8d84512ff1982330a684bad920ec4262fe

Kashu7100, 2025-01-25:
Can you check https://github.com/Genesis-Embodied-AI/Genesis/issues/352#issuecomment-2564272924 ?

irrikrlla, 2025-01-27:
> Can you check [#352 (comment)](https://github.com/Genesis-Embodied-AI/Genesis/issues/352#issuecomment-2564272924) ?

Thanks so much, I checked libglib2.0-0, libegl1, libgles2 in my wsl,  and add os.environ['PYOPENGL_PLATFORM'] = 'glx' before import genesis. Problem solved

closed 2025-01-27 by Kashu7100