Genesis-Embodied-AI/genesis-world #637: [Bug]: Can't load and render big mjcf scene

Felix-Zhenghao, 2025-01-26:
### Bug Description

# Issue
I am currently trying to load robocasa scene into genesis. However, when I directly import the mjcf of the scene, the process is just kill. My guess is the RAM is not enough to run this big scene. Is there any potential ways to solve this? The error:

```
[Genesis] [19:06:41] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [19:06:41] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [19:06:41] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [19:06:41] [INFO] Running on [NVIDIA GeForce RTX 3070 Ti Laptop GPU] with backend gs.cuda. Device memory: 7.68 GB.
[Genesis] [19:06:42] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [19:06:42] [INFO] Scene <1606103> created.
[Genesis] [19:06:42] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <b0122bd>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [19:06:42] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <c171b85>, morph: <gs.morphs.MJCF(file='/home/boris/Desktop/robocasa/layout_0_style_1.xml')>, material: <gs.materials.Rigid>.
Killed
```

# SYSTEM:
RTX 3070 ti, 8GB 
Ubuntu 20.04

### Steps to Reproduce

```
import genesis as gs
import numpy as np

gs.init(backend=gs.gpu)

scene = gs.Scene(
    show_viewer = True,
)

plane = scene.add_entity(
    gs.morphs.Plane()
)
robocasa = scene.add_entity(
    gs.morphs.MJCF(file='/home/boris/Desktop/robocasa/layout_0_style_1.xml')
    
)

scene.build()

for i in range(1200):
    scene.step()
```

### Expected Behavior

See the scene in UI rather than process killed.

### Relevant log output

```shell
[Genesis] [19:06:41] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [19:06:41] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [19:06:41] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [19:06:41] [INFO] Running on [NVIDIA GeForce RTX 3070 Ti Laptop GPU] with backend gs.cuda. Device memory: 7.68 GB.
[Genesis] [19:06:42] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [19:06:42] [INFO] Scene <1606103> created.
[Genesis] [19:06:42] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <b0122bd>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [19:06:42] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <c171b85>, morph: <gs.morphs.MJCF(file='/home/boris/Desktop/robocasa/layout_0_style_1.xml')>, material: <gs.materials.Rigid>.
Killed
```

### Environment

- OS: Ubuntu 20.04
- GPU/CPU RTX 3070 Ti
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version 12

### Release version or Commit ID

Genesis 0.21

Felix-Zhenghao, 2025-01-26:
I also tired to turn off the render, but still didn't work.

bxtbold, 2025-01-26:
@Felix-Zhenghao Can you upload the xml file? I will take a look at it

Felix-Zhenghao, 2025-01-26:
Hi, sorry the xml is too long and github does not allow me to upload here.

My email is felixchi422@gmail.com           I could send you this xml file.

Felix-Zhenghao, 2025-01-27:
# More information of reproduction

To reproduce, you may need to download robocasa and its assets by following [this page.](https://robocasa.ai/docs/introduction/installation.html) To get the scene xml, you should add these lines in the `demo_kitchen_scenes.py` before [`ep_directory, discard_traj = collect_human_trajectory` ](https://github.com/robocasa/robocasa/blob/main/robocasa/demos/demo_kitchen_scenes.py#L157C9-L157C62):

```
        with open("layout_5.xml",'w') as f:
            f.write(env.sim.model.get_xml())
```

Then run this command: 
```
python robocasa/robocasa/demos/demo_kitchen_scenes.py --task PrepareToast
```

This will give you a mjcf file that I get.

closed 2025-02-04 by Felix-Zhenghao