Genesis-Embodied-AI/genesis-world #647: [Bug]:  % python examples/locomotion/go2_train.py

malv-c, 2025-01-28:
### Bug Description

(gw) /2/5/git/Genesis
% python examples/locomotion/go2_train.py 
Traceback (most recent call last):
  File "/2/5/git/Genesis/examples/locomotion/go2_train.py", line 7, in <module>
    from rsl_rl.runners import OnPolicyRunner
ModuleNotFoundError: No module named 'rsl_rl'

### Steps to Reproduce

(gw) /2/5/git/Genesis
% python examples/locomotion/go2_train.py 
Traceback (most recent call last):
  File "/2/5/git/Genesis/examples/locomotion/go2_train.py", line 7, in <module>
    from rsl_rl.runners import OnPolicyRunner
ModuleNotFoundError: No module named 'rsl_rl'

### Expected Behavior

idk

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

git latest

fzargarbashi, 2025-01-28:
In your environment, install rsl_rl:
```
git clone https://github.com/leggedrobotics/rsl_rl
cd rsl_rl
pip install -e .
```

malv-c, 2025-01-28:
THANKS
now:
ModuleNotFoundError: No module named 'tensorboard'
can you make  pip install all deps  please ?
 it's for everyone ;)

YilingQiao, 2025-02-02:
Thanks @fzargarbashi for answering the question! 

Genesis can run with CPU only. Maybe you have some other packages that need CUDA?

malv-c, 2025-02-03:
File "/M/conda/envs/gw/lib/python3.11/site-packages/genesis/utils/misc.py", line 117, in get_gpu_device
    gs.raise_exception("cuda device not available")
Genesis can run with CPU only.
sorry but no.

Genesis-Embodied-AI, 2025-02-26:
if you read the doc - you can specify the backend device

closed 2025-02-26 by Genesis-Embodied-AI