Genesis-Embodied-AI/genesis-world #649: [Feature]: Remove debug lines

benjamin-levin, 2025-01-28:
### What feature or enhancement are you proposing?

I want to be able to remove draw_debug_line entities that I have added to the scene.

### Motivation

I was using debug lines to draw target paths and real paths for my robot end-effectors in the environment. After some time tracking the end-effectors real movements, many line objects accumulate in the scene and the simulation slows down. I want to be able to remove old line entities so there is only a recent history of movements represented.

### Potential Benefit

It will make it easier to dynamically add and remove elements from scenes. 

### What is the expected outcome of the implementation work?

I should be able to store line_entity objects returned by self.scene.draw_debug_line() and remove them using self.scene.remove_entity(line_entity). The line should then disappear from the scene and the simulation should become lighter.

Kashu7100, 2025-01-29:
I think you can do this by `scene.clear_debug_objects()`

benjamin-levin, 2025-01-29:
This would clear all objects, not the specific one I want to clear. I want the end effector to leave a disappearing trail of where it has been. I was able to modify the module code to implement this functionality. I would love to share implementation so that it could be added to the codebase.

Kashu7100, 2025-01-30:
I see! PR is welcome!!

benjamin-levin, 2025-01-30:
Awesome! I will refactor the rest of the debug functions to integrate the ability to remove specific debug entities across the entire set. 

At the moment the trailing feature I created, creates line segments at a predefined rate which isn't super efficient for slow straight-line movements. I will add a smarter fully customizable implementation of the trail drawing feature where it will automatically merge connected line-segments that form a sufficiently straight line so it minimizes added elements. I will submit a PR request with all these features once it's completed.

bxtbold, 2025-01-31:
hey @benjamin-levin. I have been working on this feature, and it seems like you have been working on this too! If possible, please take a look at #660 and give ur opinions and/or reviews!!

YilingQiao, 2025-02-02:
Thanks for @bxtbold . It's now merged and you can refer to
```
examples/tutorials/draw_debug.py
```

closed 2025-02-02 by YilingQiao