Genesis-Embodied-AI/genesis-world #650: [Bug]: Apple M3: Viewer lookat in top right corner

alexis779, 2025-01-29:
### Bug Description

The lookat position `(0.0, 0.0, 0.5)` is located on the top right corner of the window.

### Steps to Reproduce

Script is similar to the one in `./examples/render_on_macos.py`.

### Script

```python
import genesis as gs

def main():
    gs.init(backend=gs.cpu)

    lookat = (0.0, 0.0, 0.5)

    scene = gs.Scene(
        show_FPS=False,
        viewer_options=gs.options.ViewerOptions(
            camera_lookat=lookat,
        ),
    )

    cam = scene.add_camera(lookat=lookat)

    scene.add_entity(gs.morphs.Plane())

    scene.add_entity(
        gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
    )

    scene.build()

    cam.start_recording()

    gs.tools.run_in_another_thread(fn=run_sim, args=(scene, cam, ))
    scene.viewer.start()

def run_sim(scene, cam):
    for _ in range(100):
        scene.step()
        cam.render()
    
    cam.stop_recording(save_to_filename='video.mp4')

if __name__ == "__main__":
    main()

```

### Expected Behavior

The viewer should center the lookat position in the middle of the window.

### Screenshots/Videos

[image]

The viewer recording displays the robot in the top right corner of the window
https://github.com/user-attachments/assets/0eecdff5-0135-4485-8830-20ad4d1c1684

The camera recording does not show the issue.
https://github.com/user-attachments/assets/b1c13bd8-a432-4efa-90ad-df930f051a1b

### Relevant log output

```shell
% python ./examples/render_on_macos.py
[Genesis] [18:21:34] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [18:21:34] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [18:21:34] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [18:21:34] [INFO] Running on [Apple M3 Max] with backend gs.cpu. Device memory: 64.00 GB.
[Genesis] [18:21:34] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [18:21:34] [INFO] Scene <e83dd07> created.
[Genesis] [18:21:34] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <9cf7f86>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [18:21:34] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <f1e4837>, morph: <gs.morphs.MJCF(file='/Users/adetreglode/Documents/python/robotics/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [18:21:35] [INFO] Building scene <e83dd07>...
[W 01/28/25 18:21:35.203 47395369] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/28/25 18:21:35.360 47395369] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/28/25 18:21:36.457 47395369] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [18:21:36] [INFO] Compiling simulation kernels...
[Genesis] [18:21:38] [INFO] Building visualizer...
[Genesis] [18:21:38] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`.
[Genesis] [18:21:38] [INFO] Viewer created. Resolution: 744×558, max_FPS: 60.
UNSUPPORTED (log once): POSSIBLE ISSUE: unit 4 GLD_TEXTURE_INDEX_CUBE_MAP is unloadable and bound to sampler type (Float) - using zero texture because texture unloadable
[Genesis] [18:21:41] [INFO] Saving video to video.mp4.
[Genesis] [18:21:41] [INFO] Video saved.
[Genesis] [18:26:50] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

### Environment

- OS: Apple M3
- GPU/CPU cpu
- GPU-driver version: N/A
- CUDA / CUDA-toolkit version: ### Logs

### Release version or Commit ID

7e0adc03c424838550f8c1fb6f4c1f814bde5ae6

herc8911, 2025-02-03:
I got same errors on after I re-install the whole envirement, before it works.

[image]

[image]
[image]

herc8911, 2025-02-03:
Set if sys.platform = "darwin": self.dpscale=1 in render.py.

[image]

[image]

[image]

alexis779, 2025-02-03:
Thanks! Setting `self.dpscale = 1` fixes the issue.

Was the 2x scaling required for Apple M1 & M2 processors ?

On M3 it's no longer required. `get_cpu_name` in `misc.py` will return Apple CPU type, so we could check it besides `darwin` platform.

```
% sysctl -n machdep.cpu.brand_string
Apple M3 Max
```

neil-tan, 2025-02-11:
An alternative is to add this after scene.build() and before scene.viewer.start():
```python
import sys
if sys.platform == "darwin":
  scene._visualizer._viewer._pyrender_viewer._renderer.dpscale = 1
```

closed 2025-02-26 by duburcqa