Genesis-Embodied-AI/genesis-world #653: [Bug]: A NaN error occasionally occurs in get_pos() when colliding with the ground.

trhrobo, 2025-01-29:
### Bug Description

I am currently working on locomotion learning for a robot with multiple contact points, and I am facing an issue where get_pos() returns NaN upon reset. I applied the suggested fix from #625, which is setting convexify=False, but the problem was not resolved.

[image]

What is strange is that this issue does not occur every time; rather, it happens at some point during training after some progress has been made. Even though the reset position is the same for each episode, some episodes proceed without any NaN errors, while occasionally NaN occurs. For example, in the attached image, NaN occurred at iteration 84/100, causing the training to stop. The iteration where NaN occurs varies depending on the environment. When running the training on another PC, NaN appears at a different iteration.

[image]

By visualizing the training process when NaN occurs, I found that it happens when the robot is reset in free space and lands on or approaches the ground.

Could you help me identify the cause of this issue and suggest a solution?

### Steps to Reproduce

self.robot is defined as shown below (even if convexity=False, an error occurs).
```python
 self.robot = self.scene.add_entity(
       gs.morphs.URDF(
                file=self.cfg.asset.file,
                merge_fixed_links=True,
                links_to_keep=self.cfg.asset.links_to_keep,
                pos=self.base_init_pos.cpu().numpy(),
                quat=self.base_init_quat.cpu().numpy(),
                convexify=True,
         ),
         visualize_contact=self.debug,
 )
```
in simulation, I get robot base position using get_pos()
```python
self.base_pos[:] = self.robot.get_pos()
```
I have found that NaN occurs when the robot approaches the ground.

```python
if torch.isnan(self.obs_buf).any():
       nan_indices = torch.nonzero(torch.isnan(self.obs_buf), as_tuple=False).squeeze()
       print(f"now_step: {self.base_pos[nan_indices[0]]}")
       print(f"prev_1step: {self.prev_base_pos[nan_indices[0]]}")
       print(f"prev_2step: {self.prev_prev_base_pos[nan_indices[0]]}")
       print(f"prev_3step: {self.prev_prev2_base_pos[nan_indices[0]]}")
       print(f"prev_4step: {self.prev_prev3_base_pos[nan_indices[0]]}")
```

[image]

### Expected Behavior

Continue training without outputting NaN.

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 22.04]
- GPU/CPU [GeForce RTX3060 Mobile] 
- GPU-driver version 12.1
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

v0.2.1

Kashu7100, 2025-01-29:
Can you try to reduce the simulation dt to see what happens? You can check `SimOptions` and `RigidOptions` for this.

trhrobo, 2025-01-30:
@Kashu7100 
The original code was running at 50Hz (with a decimation factor of 6). I tried increasing the frequency to 80Hz and 100Hz (reducing dt), but the issue persists, and NaN still occurs. However, the iteration at which NaN appears changes: at 80Hz, it occurred at iteration 929, while at 100Hz, it occurred at iteration 51.
```python
    class control:
        # PD
        PD_stiffness = {'Revolute': 80.0} #80[N*m/rad]
        PD_damping = {'Revolute': 3.0} #1[N*m*s/rad]
        action_scale = 0.25
        # decimation: Number of control action updates @ sim DT per policy DT
        decimation = 6
        control_freq = 80 #[Hz]
        use_implicit_controller = False
```
```python
        self.dt = 1 / self.cfg.control.control_freq
        self.sim_dt = self.dt / self.cfg.control.decimation
        if self.cfg.control.use_implicit_controller:
            sim_dt = self.dt
            sim_substeps = self.cfg.control.decimation
        else:
            sim_dt = self.dt / self.cfg.control.decimation
            sim_substeps = 1
```
```python
        self.scene = gs.Scene(
            sim_options=gs.options.SimOptions(
                dt=sim_dt,
                substeps=sim_substeps,
            ),
            viewer_options=gs.options.ViewerOptions(
                max_FPS=int(1 / self.dt * self.cfg.control.decimation),
                camera_pos=(2.0, 0.0, 2.5),
                camera_lookat=(0.0, 0.0, 0.5),
                camera_fov=40,
            ),
            vis_options=gs.options.VisOptions(
                n_rendered_envs = 1 if self.num_envs == 1 else 8
            ),
            rigid_options=gs.options.RigidOptions(
                dt=sim_dt,
                constraint_solver=gs.constraint_solver.Newton,
                enable_collision=True,
                enable_self_collision=True,
                enable_joint_limit=True,
            ),
            show_viewer=show_viewer,
            show_FPS=False,
        )
```
80Hz error
[image]
[image]

100Hz error
[image]
[image]

Kashu7100, 2025-01-30:
Can you try `sim_dt=0.005` and `decimation=4`, which I use for RL often?

trhrobo, 2025-01-30:
Even after changing sim_dt=0.005 and decimation=4, the issue still persists.
[image]

zhouxian, 2025-01-31:
@YilingQiao could you take a look?

trhrobo, 2025-01-31:
I conducted a more detailed investigation of the bug. I set the decimation to 4 and examined how each joint (my robot has 34 joints) behaves in the world frame at every self.scene.step(). The image tensor stores the values of xyz for each joint, with joints arranged vertically. Strangely, at the first instance where NaN appears, base_link does not contain NaN, while NaN occurs in other parts.
[image]

I also confirmed that the actions before and after the occurrence of NaN did not take any particularly unusual values. This control mechanism specifies torque for each joint using self.robot.control_dofs_force().
The image below represents the torque values (actions) applied to each joint.
[image]

trhrobo, 2025-02-01:
I have created a simplified repository where the bug can be observed.
https://github.com/trhrobo/gs_snake

When executing the code from this repository, the result appears as shown here.
[image]
[image]

YilingQiao, 2025-02-01:
thank you! I will take a look

YilingQiao, 2025-02-02:
I can run your script and replicate this issues. Will let you know when I make progress.

YilingQiao, 2025-02-02:
I found that it's NAN because the velocity is too large and so explode
```

    def step(self, actions):
        self.actions = torch.clip(actions, -self.env_cfg["clip_actions"], self.env_cfg["clip_actions"])
        exec_actions = self.last_actions if self.simulate_action_latency else self.actions
        target_dof_pos = exec_actions * self.env_cfg["action_scale"] + self.default_dof_pos
        self.robot.control_dofs_position(target_dof_pos, self.motor_dofs)
        
        rigid = self.scene.sim.rigid_solver
        
        for i in range(4):
            torques = self._compute_torques(exec_actions)
            self.robot.control_dofs_force(torques, self.motor_dofs)
            self.scene.step()

            qpos = rigid.qpos.to_numpy()
            qvel = rigid.dofs_state.vel.to_numpy()
            

            if np.isnan(qpos).any():
                idx_nan = np.argwhere(np.isnan(qpos).any(0))[0, 0]
                last_qpos = self.last_qpos[:,idx_nan]
                last_qvel = self.last_qvel[:,idx_nan]
                last_torques = torques[idx_nan,:]
                print("last_qvel", last_qvel)
                print("last_torques", last_torques)
                from IPython import embed; embed()
            self.last_torques = torques.clone()
            self.last_qpos = qpos
            self.last_qvel = qvel

```
The torque and qvel of the exploded environment are significantly larger than other envs.
```
last_qvel [ 5.3885281e+09 -4.3296635e+09 -1.0083014e+10  3.8498951e+10
 -7.0443147e+10  1.1265249e+11 -1.3130841e+10  4.8093000e+07
 -1.6759799e+09 -3.5982275e+08  2.6685530e+08  1.1925520e+11
  4.5649270e+08 -4.1970614e+10 -5.6635046e+08  1.0500044e+11
  4.4488931e+08  2.3971574e+09 -4.1403469e+08 -5.8026726e+08
  3.4645540e+09  1.8941480e+11 -4.9812720e+07 -2.7760637e+11
  2.5998072e+09  5.2572049e+11 -1.2196088e+09  8.0254227e+08
 -7.7995923e+08  4.1741737e+09 -1.6379234e+11 -5.1393793e+11
  4.9696707e+09  2.7002792e+11  1.2272375e+11 -4.2343346e+10
 -1.5465537e+09  1.9574829e+08 -7.4365434e+08 -3.6967370e+09]
last_torques tensor([ 100.0000, -100.0000,  100.0000,  100.0000,  100.0000, -100.0000,
        -100.0000,   93.6800,  100.0000, -100.0000, -100.0000,  100.0000,
        -100.0000, -100.0000, -100.0000,  100.0000,  100.0000, -100.0000,
         -80.2339, -100.0000,  100.0000,  100.0000, -100.0000, -100.0000,
         100.0000,  100.0000,  100.0000, -100.0000, -100.0000,  100.0000,
         100.0000, -100.0000,  100.0000, -100.0000], device='cuda:0')

```

YilingQiao, 2025-02-02:
Can you penalize the actuation magnitude more in your reward function? Or reset/early terminate NAN envs and give them a big penalty?

YilingQiao, 2025-02-02:
I found that you use
```
        for i in range(4):
            torques = self._compute_torques(exec_actions)
            self.robot.control_dofs_force(torques, self.motor_dofs)
            self.scene.step()

```
It's actually applying the same action for 4 dt, so the actual control frequency is 50/4 = 12.5?

If I understood correctly, you should change this 4->1, while using bigger substeps, like
```
sim_options=gs.options.SimOptions(dt=self.dt, substeps=5),
```

By doing this, the control frequency stays the same while simulation is more accurate.

With this setting, the simulation will not blow up on my side

Kashu7100, 2025-02-02:
```python
 for i in range(4):
            torques = self._compute_torques(exec_actions)
            self.robot.control_dofs_force(torques, self.motor_dofs)
            self.scene.step()
```
@YilingQiao I think this setup is common in Isaac code base for locomotion. Maybe we need to emphasis this in doc. 
@ziyanx02 Any thoughts?

YilingQiao, 2025-02-02:
I see. Thanks! Can you or @ziyanx02 add this to the doc?

Kashu7100, 2025-02-02:
@trhrobo @YilingQiao Adding the updates of `dof_pos` and `dof_vel` within the decimation steps works (pointed out by @ziyanx02 ). 

[image]

trhrobo, 2025-02-03:
@Kashu7100 @YilingQiao @zhouxian 
Thank you for solving the problem! The result turned out exactly as I had hoped!

closed 2025-02-03 by trhrobo