Genesis-Embodied-AI/genesis-world #655: [Bug]: AttributeError: 'NoneType' object has no attribute 'set_current'

zhjc2014, 2025-01-30:
### Bug Description

> Exception in thread Thread-2:
> Traceback (most recent call last):
>   File "D:\ProgramData\miniconda3\envs\genesis\lib\threading.py", line 980, in _bootstrap_inner
>     self.run()
>   File "D:\ProgramData\miniconda3\envs\genesis\lib\threading.py", line 917, in run
>     self._target(*self._args, **self._kwargs)
>   File "D:\myfile\src\Genesis\genesis\ext\pyrender\viewer.py", line 1169, in _init_and_start_app
>     self.switch_to()
>   File "D:\ProgramData\miniconda3\envs\genesis\lib\site-packages\pyglet\window\win32\__init__.py", line 385, in switch_to
>     self.context.set_current()
> AttributeError: 'NoneType' object has no attribute 'set_current'

when running `python .\examples\drone\fly.py -v`

### Steps to Reproduce

`python .\examples\drone\fly.py -v`

### Expected Behavior

no Exception

### Relevant log output

```shell

```

### Environment

- OS: [Windows 11 21H2]
- GPU/CPU [RTX 4090, Intel I9-12900kf]
- GPU-driver version 571.96
- CUDA / CUDA-toolkit version 12.8

### Release version or Commit ID

Commit: 7e0adc03c424838550f8c1fb6f4c1f814bde5ae6
Parents: 806d0a8d84512ff1982330a684bad920ec4262fe, bf9f31ad751a1ecd90e4b06a722b35fbb6cb9570
Author: Kashu Yamazaki <[39104615+Kashu7100@users.noreply.github.com](mailto:39104615+Kashu7100@users.noreply.github.com)>
Committer: GitHub <[noreply@github.com](mailto:noreply@github.com)>
Date: Mon Jan 27 2025 06:27:59 GMT+0800 (China Standard Time)

bxtbold, 2025-01-31:
There's an error in the pyglet not in genesis. Maybe you should try installing [the latest pyglet release](https://pypi.org/project/pyglet/#history).

duburcqa, 2025-03-23:
Could you try with the main brand? I think this issue has been fixed or at least the traceback should be much clearer.