Genesis-Embodied-AI/genesis-world #671: [Bug]:

4EverBuilder, 2025-02-04:
### Bug Description

Getting "Illegal instruction (core dumped)" when importing genesis

```
python
Python 3.12.7 (main, Jan 17 2025, 16:55:27) [GCC 14.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import genesis as gs
Illegal instruction (core dumped)
```

machine type

```
Linux desktop1 6.11.0-14-generic #15-Ubuntu SMP PREEMPT_DYNAMIC Fri Jan 10 23:48:25 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

### Steps to Reproduce

Open Python and run import genesis as gs

### Expected Behavior

Should import without issue?

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 24.04
- Intel(R) Xeon(R) CPU E5620  @ 2.40GHz 
- No GPU
- CUDA / CUDA-toolkit version N/A

### Release version or Commit ID

latest

zhouxian, 2025-02-06:
Could you downgrade to say python 3.10 to see if it persists?

4EverBuilder, 2025-02-11:
It presists even with docker. What can I share with you?

zhouxian, 2025-02-12:
Is it possible to set breakpoints inside the code and see which exact line
is causing this? It’s pretty difficult to reproduce this from our side

4EverBuilder, 2025-02-12:
It was failing on mujoco
Found out it's because of mujoco wanting AVX2 instruction. Recompliling mojuco without avx worked!

`cmake .  -DCMAKE_INSTALL_PREFIX=/usr/local/mujoco/  -DCMAKE_CXX_FLAGS="-march=nehalem -mtune=generic -mno-avx -mno-avx2 -mno-sse4.2 -mno-fma -mno-avx512f"`

closed 2025-02-12 by 4EverBuilder

youdass101, 2025-02-15:
Same issue here, using Oracle VM virtualBox. 
clean ubuntu installation, using python 3.12.8

FanMover, 2025-02-25:
Any idea about Mac mini M4 docker genesis? Illegal instruction when import genesis.

duburcqa, 2025-02-25:
Could you all check whether it is related to Mujoco or Genesis itself ? Just try doing:
```python
import mujoco
```

Gbrothers1, 2025-04-16:
> It was failing on mujoco Found out it's because of mujoco wanting AVX2 instruction. Recompliling mojuco without avx worked!
> 
> `cmake . -DCMAKE_INSTALL_PREFIX=/usr/local/mujoco/ -DCMAKE_CXX_FLAGS="-march=native -mtune=generic -mno-avx -mno-avx2 -mno-sse4.2 -mno-fma -mno-avx512f -mno-bmi -mno-bmi2"`

Would you mind clarifying on this, does this change need to occur in the docker file or can I just patch and rebuild Mujoco from within the container. I am one of those disturbed individuals without an AVX supported CPU unfortunately. Will they ever update the installation instructions for non-avx CPU's ?

4EverBuilder, 2025-07-29:
> Would you mind clarifying on this, does this change need to occur in the docker file or can I just patch and rebuild Mujoco from within the container. I am one of those disturbed individuals without [...]

Issue is with Mujoco. I think they have stopped supporting nonAVX completely per latest mujoco :(