Genesis-Embodied-AI/genesis-world #672: [Bug]: Very low FPS when running vis_mode='recon'

sickopickle, 2025-02-04:
### Bug Description

I am running the code provided in the tutorial "Beyond Rigid Bodies". I modified my code to use vis_mode='recon':

`import genesis as gs

########################## init ##########################
gs.init()

########################## create a scene ##########################

scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt       = 4e-3,
        substeps = 10,
    ),
    sph_options=gs.options.SPHOptions(
        lower_bound   = (-0.5, -0.5, 0.0),
        upper_bound   = (0.5, 0.5, 1),
        particle_size = 0.01,
    ),
    vis_options=gs.options.VisOptions(
        visualize_sph_boundary = True,
    ),
    show_viewer = True,
)

########################## entities ##########################
plane = scene.add_entity(
    morph=gs.morphs.Plane(),
)

liquid = scene.add_entity(
    # viscous liquid
    # material=gs.materials.SPH.Liquid(mu=0.02, gamma=0.02),
    material=gs.materials.SPH.Liquid(),
    morph=gs.morphs.Box(
        pos  = (0.0, 0.0, 0.65),
        size = (0.4, 0.4, 0.4),
    ),
    surface=gs.surfaces.Default(
        color    = (0.4, 0.8, 1.0),
        vis_mode = 'recon',
    ),
)

########################## build ##########################
scene.build()

horizon = 1000
for i in range(horizon):
    scene.step()

# get particle positions
particles = liquid.get_particles()`

With the default code it runs fine but the FPS is <0.5. I have a 4070 Ti.

### Steps to Reproduce

```
import genesis as gs

########################## init ##########################
gs.init()

########################## create a scene ##########################

scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt       = 4e-3,
        substeps = 10,
    ),
    sph_options=gs.options.SPHOptions(
        lower_bound   = (-0.5, -0.5, 0.0),
        upper_bound   = (0.5, 0.5, 1),
        particle_size = 0.01,
    ),
    vis_options=gs.options.VisOptions(
        visualize_sph_boundary = True,
    ),
    show_viewer = True,
)

########################## entities ##########################
plane = scene.add_entity(
    morph=gs.morphs.Plane(),
)

liquid = scene.add_entity(
    # viscous liquid
    # material=gs.materials.SPH.Liquid(mu=0.02, gamma=0.02),
    material=gs.materials.SPH.Liquid(),
    morph=gs.morphs.Box(
        pos  = (0.0, 0.0, 0.65),
        size = (0.4, 0.4, 0.4),
    ),
    surface=gs.surfaces.Default(
        color    = (0.4, 0.8, 1.0),
        vis_mode = 'recon',
    ),
)

########################## build ##########################
scene.build()

horizon = 1000
for i in range(horizon):
    scene.step()

# get particle positions
particles = liquid.get_particles()
```

### Expected Behavior

I expect the simulation to run at 60+ FPS

### Screenshots/Videos

[image]

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 22.04
- GPU/CPU: RTX 4070 Ti, 12900k
- GPU-driver version 535
- CUDA / CUDA-toolkit version 12.2

### Release version or Commit ID

0.2.1

lgleim, 2025-02-05:
Are you implicitly running with CPU backend? Cf. https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html#initialization

lgleim, 2025-02-05:
Surfance reconstruction is done via a CPU backend. Cf. https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html#optional-surface-reconstruction. Thus you are CPU bound and your GPU mostly idles. To overcome this, a TaiChi implementation of splashsurf / particlemesher or a similar approach would have to be implemented / integrated.

w1th0utnam3, 2025-02-06:
Author of [splashsurf](https://github.com/InteractiveComputerGraphics/splashsurf) here. In the short term we cannot do much about making it run even nearly in realtime for such amount of particles and grid resolutions (~60k particles resulting in 120k triangles in 170ms-190ms). I would say it is quite well parallelized on the _CPU_ (no explicit vectorization though) but doesn't make any use of the GPU and therefore cannot compare with the performance of e.g. a GPU implementation of SPH itself. Of course there are also some design decisions in splashsurf that are relevant for performance, for example we do not extract only surface particles but we work with all particles so that the we do not miss any cavities or produce holes in the mesh.
 
A very quick fix to _slightly_ improve performance on the side of Gensis is btw. to let splashsurf write VTK or PLY files and not OBJ (if supported by the mesh loading) because those are 10x faster than the ASCII writer for OBJ. Benchmarking this scene, the OBJ writing is currently ~20% of the entire runtime of the splashsurf call. See [particle.py#L319](https://github.com/Genesis-Embodied-AI/Genesis/blob/2207bed8fdee65eb692aecaa93d0b01dd0d8b961/genesis/utils/particle.py#L319).

Of course, ideally no disk IO would be required at all using a Python module for splashsurf and we might offer this in the future.

We mostly use splashsurf for offline reconstruction of simulations and as such we did not have the need for interactive performance. It would be interesting though to explore GPU implementations in the future as @lgleim suggested.

closed 2025-05-24 by duburcqa