Genesis-Embodied-AI/genesis-world #681: [Feature]:  Rigid assemble

Xu-Xuanhui, 2025-02-06:
### What feature or enhancement are you proposing?

 I want to combine two rigid bodies into one when certain conditions are met during simulation, and then separate them when they are not met?

### Motivation

As a [e.g. Embodied intelligence researcher working on space robot]
I want to implement in the simulation process that robots use suction cups to crawl on the surface of spacecraft.
I want the code interface to be established by creating joints between the robot's feet and the surface of the spacecraft.

### Potential Benefit

This function can be widely used in applications such as assembly and suction cups.

### What is the expected outcome of the implementation work?

- I want the code interface to be established by creating joints between the robot's feet and the surface of the spacecraft.

Kashu7100, 2025-02-06:
Yes, we need this. 

For now a workaround would be that you manually control the object by `control_dofs_position` to follow the end-effector

ArthurKaede, 2025-02-20:
If you use `control_dofs_position`, how can you ensure that the end-effector remains fixed in a certain position?

YilingQiao, 2025-03-31:
Added in #961

closed 2025-03-31 by YilingQiao