Genesis-Embodied-AI/genesis-world #682: [Feature]: Marine and wave simulation, buoyancy, hydrodynamics

sickopickle, 2025-02-06:
### What feature or enhancement are you proposing?

Are there any plans for wave simulation [like this](https://github.com/srmainwaring/asv_wave_sim)? Can I somehow use Gazebo alongside Genesis to use this marine environment? I'm aware there is particle fluid simulation, but I am looking for something on a more macro scale (like with trochoids, as shown in [this slideshow](https://docs.google.com/presentation/d/1JXwWMPPVT7y03Vr6cWtrwAwyvdysmg3NW9ZmI276dMY/edit#slide=id.g1343670f3e0_0_37)) rather than simulating the individual particles. 

### Motivation

I am trying to train a reinforcement learning algorithm for drones to land on boats (compensating for rocking from waves). 

### Potential Benefit

Opens up a lot of possibilities for marine simulation and applications there.

### What is the expected outcome of the implementation work?

- A marine environment with adjustable parameters
- Waves
- Buoyancy
- Hydrodynamics
- Different boat models
- Hopefully GPU-accelerated (I noticed that the splashsurf surface reconstruction does not run on GPU)

Kashu7100, 2025-02-06:
This is a lower priority for us at the moment, as we're still focused on implementing and refining fundamental functionalities. However, if you're familiar with this type of simulation, we always welcome PRs!
If I understand correctly, a separate solver would be needed to simulate waves on a macro scale.

sickopickle, 2025-02-06:
I don't have much experience in actually implementing it, but would it be possible to integrate the existing Gazebo plugin I've linked above? One issue is that it is implemented in C++, whereas Genesis is Python

duburcqa, 2026-11-12:
We now have a high-quality fluid simulator based on DFSPH, which includes wave simulation, buoyancy, and hydrodynamics. It supports coupling with rigid and FEM entities.

closed 2026-11-12 by duburcqa