Genesis-Embodied-AI/genesis-world #685: [Bug]: SPH particles scatter uncontrollably upon collision

Naoki04, 2025-02-06:
### Bug Description

In the SPH fluid simulation, when particles collide with a mesh or the floor, they scatter at an extremely high velocity. This issue persists regardless of the collision surface. While reducing the collision velocity to a very small value or decreasing the particle size mitigates the issue, these adjustments are not feasible for our current water-pouring task. The problem occurs consistently and severely impacts the stability of the simulation.

### Steps to Reproduce

	1.	Set up an SPH fluid simulation with a pouring task.
	2.	Allow particles to collide with either a mesh object or the ground.
	3.	Observe that particles scatter uncontrollably at high speed.
	4.     You will find the problem mitigation when setting particle_size=0.01

```
import genesis as gs
import time
import torch
import numpy as np

dt = 4e-3
########################## init ##########################
gs.init(
    seed                = None,
    precision           = '32',
    debug               = False,
    eps                 = 1e-12,
    logging_level       = None,
    backend             = gs.gpu,
    theme               = 'dark',
    logger_verbose_time = False,
    
)

########################## Create a scene ##########################
scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt       = dt,
        substeps = 10,
    ),
    sph_options=gs.options.SPHOptions(
        lower_bound   = (-5, -5, 0.0),
        upper_bound   = (5, 5, 10),
        particle_size = 0.004, # Default is 0.01
    ),
    vis_options=gs.options.VisOptions(
        visualize_sph_boundary = False,
        show_world_frame = False,
        shadow = False,
    ),
    show_viewer = False,
)

########################## Scene Setup ##########################
plane = scene.add_entity(gs.morphs.Plane())

# Add the liquid
liquid = scene.add_entity(
    material=gs.materials.SPH.Liquid(
        sampler='pbs',
    ),
    morph=gs.morphs.Cylinder(
        radius=0.035, 
        height=0.10, 
        pos=(0,0.03,0.05)
    ),
    surface=gs.surfaces.Default(
        color    = (0.4, 0.8, 1.0),
        vis_mode = 'particle',
    ),
)

# Add the camera
cam = scene.add_camera(
    res    = (1280, 960),
    pos    = (0.6, -0.4, 0.2),
    lookat = (0, 0, 0.1),
    fov    = 30,
    GUI    = False
)

"""
Run the simulation
"""
scene.build()

cam.start_recording()
steps = 2000
t = time.time()
for i in range(steps):
    scene.step()
    cam.render()

    
cam.stop_recording(save_to_filename='bug_ok.mp4', fps=int(1/dt))
```

### Expected Behavior

Particles should remain stable upon collision, exhibiting natural dispersion and not gaining extreme velocities.

### Screenshots/Videos

particle_size=0.004: https://youtu.be/CGun7oyN1JY
particle_size=0.01: https://youtu.be/Om9kNhfwgHM

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 22.04
- GPU RTX 4090

### Release version or Commit ID

genesis-world            0.2.1

Kashu7100, 2025-02-07:
This looks like a simulation instability. You need a smaller dt or a bigger substeps. I can confirm increasing the substeps to `20` works with `particle_size = 0.004`.

Kashu7100, 2025-02-07:
Original:
[image]

`particle_size = 0.004` and `substeps=20`
[image]

Naoki04, 2025-02-07:
Wow, I see. That's the point.
Thank you for your quick reply!

closed 2025-02-07 by Naoki04