Genesis-Embodied-AI/genesis-world #690: [Bug]: Incorrect Collision Handling Between Robot and Mesh in Genesis

hs-black, 2025-02-08:
### Bug Description

I am experiencing an issue with the collision detection between the robot and the mesh in Genesis. It appears that Genesis is not correctly handling the collision, causing the robot to penetrate the mesh. As a result, the force calculations for the robot are also incorrect.

### Steps to Reproduce

1. Unzip codes and assets in [clean_code.zip](https://github.com/user-attachments/files/18717217/clean_code.zip).
2. Run `python eval_go2.py`.
3. Observe the robot's interaction with the mesh in local dir `./debug/`. 

### Expected Behavior

The robot should correctly detect collisions with the mesh and respond appropriately without penetrating the mesh.

### Screenshots/Videos

[image]

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 24.04.1 LTS
- GPU/CPU A100-SXM4-80GB
- GPU-driver version 550.127.05
- CUDA / CUDA-toolkit version 12.4

### Release version or Commit ID

Genesis-dev：50b42c39

### Additional Context

**Steps to Reproduce:**

**Expected Behavior:**
The robot should correctly detect collisions with the mesh and respond appropriately without penetrating the mesh.

**Actual Behavior:**
The robot fails to detect the collision and penetrates the mesh, leading to incorrect force calculations.

**Code and Mesh:**

**Additional Information:**
- Genesis version: [Insert version number]
- Operating System: [Insert OS and version]
- Any other relevant information about your setup.

**Screenshots/Videos:**
If applicable, add screenshots or videos to help explain the problem.

Thank you for looking into this issue. I look forward to your response and any potential fixes or workarounds.

---

Feel free to adjust the content as needed to better fit your specific situation.

Kashu7100, 2025-02-08:
Hi, @hs-black I could not find the commit ID specified...

hs-black, 2025-02-08:
Hi, i've tried newest version 8a26cd4 just now and the problem still exists.
[image] 

> Hi, [@hs-black](https://github.com/hs-black) I could not find the commit ID specified...