Genesis-Embodied-AI/genesis-world #695: [Bug]: Unable to initialize an OpenGL 3+ context

guotong1988, 2025-02-09:
### Bug Description

try to run advanced_hybrid_robot.py on Ubuntu [CPU]

### Steps to Reproduce

```python
import numpy as np
import genesis as gs

########################## init ##########################
gs.init(seed=0, precision="32", logging_level="debug", backend=0)

######################## create a scene ##########################
dt = 3e-3
scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        substeps=10,
    ),
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(1.5, 1.3, 0.5),
        camera_lookat=(0.0, 0.0, 0.0),
        camera_fov=40,
    ),
    rigid_options=gs.options.RigidOptions(
        dt=dt,
        gravity=(0, 0, -9.8),
        enable_collision=True,
        enable_self_collision=False,
    ),
    mpm_options=gs.options.MPMOptions(
        dt=dt,
        lower_bound=(0.0, 0.0, -0.2),
        upper_bound=(1.0, 1.0, 1.0),
        gravity=(0, 0, 0),  # mimic gravity compensation
        enable_CPIC=True,
    ),
    vis_options=gs.options.VisOptions(
        show_world_frame=True,
        visualize_mpm_boundary=False,
    ),
)

########################## entities ##########################
scene.add_entity(morph=gs.morphs.Plane())

robot = scene.add_entity(
    morph=gs.morphs.URDF(
        file="urdf/simple/two_link_arm.urdf",
        pos=(0.5, 0.5, 0.3),
        euler=(0.0, 0.0, 0.0),
        scale=0.2,
        fixed=True,
    ),
    material=gs.materials.Hybrid(
        mat_rigid=gs.materials.Rigid(
            gravity_compensation=1.0,
        ),
        mat_soft=gs.materials.MPM.Muscle(  # to allow setting group
            E=1e4,
            nu=0.45,
            rho=1000.0,
            model="neohooken",
        ),
        thickness=0.05,
        damping=1000.0,
    ),
)

ball = scene.add_entity(
    morph=gs.morphs.Sphere(
        pos=(0.8, 0.6, 0.1),
        radius=0.1,
    ),
    material=gs.materials.Rigid(rho=1000, friction=0.5),
)

########################## build ##########################
scene.build()

########################## run ##########################
scene.reset()
for i in range(1000):
    dofs_ctrl = np.array(
        [
            1.0 * np.sin(2 * np.pi * i * 0.001),
        ]
        * robot.n_dofs
    )

    robot.control_dofs_velocity(dofs_ctrl)

    scene.step()

```

### Expected Behavior

-

### Screenshots/Videos

-

### Relevant log output

```shell
[Genesis] [17:39:26] [INFO] Compiling simulation kernels...
[Genesis] [17:39:34] [INFO] Building visualizer...
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/guotong/miniconda/envs/genesis/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/home/guotong/miniconda/envs/genesis/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/home/guotong/Genesis-0.2.1/genesis/ext/pyrender/viewer.py", line 1146, in _init_and_start_app
    raise ValueError("Unable to initialize an OpenGL 3+ context")
ValueError: Unable to initialize an OpenGL 3+ context
```

### Environment

- OS: Ubuntu 24.04.1 LTS
- CPU

### Release version or Commit ID

tag v0.2.1

achaosss, 2025-02-09:
I have the same question in ubuntu 24.10 (CPU)

achaosss, 2025-02-09:
When I use superuse(root) to run the code It success.

guotong1988, 2025-02-11:
@achaosss Could you please show your way in detail? Thank you!

achaosss, 2025-02-11:
> [@achaosss](https://github.com/achaosss) Could you please show your way in detail? Thank you!

When I first started, I ran the script using the root account and didn't encounter any issues. Then I switched to the user account, reinstalled environments like genesis, and problems arose. I didn't check thoroughly, but when I switched back to the root account, I could still run the script.

gonzalesMK, 2025-02-12:
I had only Nvidia GPU (no integrated Intel) and I had the same problem.

I fixed with this command:

> __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia python3 tutorials/sph_liquid.py

gonzalesMK, 2025-02-12:
I only had a single Nvidia GPU (no integrated Intel GPU) and encountered the same issue. I was able to resolve it by running the following command:

```
 __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia python3 tutorials/sph_liquid.py
```

duburcqa, 2025-03-03:
@guotong1988 Could you check whether it is now working on the main branch?

closed 2025-03-22 by duburcqa

MyToumaKazusa, 2025-06-12:
I solve this by using:
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6