Genesis-Embodied-AI/genesis-world #699: [Feature]: Ability to change the "fixed" attribute within the simulation

gojkopero, 2025-02-10:
### What feature or enhancement are you proposing?

Ability to change the "fixed" attribute of a base link of an entity on the fly. This would allow "enabling" and "disabling" objects in the scene while preserving the collisions they introduce. The same feature would be beneficial for the "collision" property as well.

### Motivation

As a robotics manipulation researcher, I would like to disable some objects in my experiments so they are not under the influence of unexpected collisions after they are manipulated.

### Potential Benefit

This feature could be crucial to many different simulations with evolving long-horizon tasks, as it could simplify the experiment itself, and maybe even reduce the computational overload.

### What is the expected outcome of the implementation work?

- Ability to change "fixed" property back and forth within the simulation loop.

ttianyuren, 2025-02-14:
Look forward to that as well!

Riften, 2025-02-20:
Is there any plan to support this feature?