Genesis-Embodied-AI/genesis-world #702: [Bug]: Cannot load material from the urdf file

JiaheXu, 2025-02-11:
### Bug Description

Hi developer.
I am loading a urdf object, I have tried Sapien, the file works but I cannot see the material(RGB color) in Genesis, please take a look. Here is the link to the urdf file[https://drive.google.com/drive/folders/1xLVwrAjz8csNC4hzVT7RBRp0kbTHM6tc?dmr=1&ec=wgc-drive-hero-goto ](url). Please take a look, thanks.

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

see color from that urdf file

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

genesis-world==0.2.1

YilingQiao, 2025-02-21:
Thanks. Looking into it.

wangyian-me, 2025-03-20:
Genesis only process texture on triangular mesh, but this is more likely to be a quad mesh. You can solve it by load it to mesh lab to convert it to a triangular mesh and them load it with genesis.

closed 2025-03-20 by wangyian-me