Genesis-Embodied-AI/genesis-world #706: [Bug]: batch实例数和线程数的关系

Happiness-Cleaner, 2025-02-11:
### Bug Description

你好！我们测试了Genesis官方名为“A Detailed Performance Benchmarking Report on Genesis”报告中的Go2程序在不同batch实例数（即envs）下运行

时的最大线程数，测试数据如上图所示。因为我们的测试结果和Genesis官方报告中的一些描述不太相符，这里有一些疑问：

您的报告中说：在Genesis中，batch中的每个环境实例严格只使用一个单线程来模拟，但是在我们的测试中Go2程序在2^9=512个batch之前最大线程

数都是47，512到16384个batch之间最大线程数为58。所以您说的线程和我们测试的最大线程数有什么不同？

[image]

### Steps to Reproduce

#!/bin/bash

# 定义最大线程数变量
max_threads=0

# 循环遍历脚本数组
for env in {12..14}; do
    # 启动 Python 脚本Go2并在后台执行
    python3 test_genesis.py -B $((2**$env)) -r --mjx-solver-setting &

    # 获取当前启动的 Python 脚本的进程 PID
    PID=$!
    
    # 输出正在运行的脚本及其 PID
    echo "Running $((2**$env)) batch with PID: $PID"
    
    # 每秒输出一次线程数，直到进程结束
    while true; do

        # 检查进程是否仍在运行
        if ! ps -p $PID > /dev/null; then
            echo "$((2**$env)) (PID $PID) has finished."
            break
        fi

        # 获取进程的线程数
        THREADS=$(ps -o nlwp= -p $PID)
        
        # 比较当前线程数和最大线程数
        if [ "$THREADS" -gt "$max_threads" ]; then
            max_threads=$THREADS
        fi

        # 等待 1 秒后再次检查
        sleep 1
    done
    
    # 输出线程数
    echo "Max Thread count for $((2**$env)) batch (PID $PID): $max_threads"

    # 重置最大线程数变量
    max_threads=0
done

### Expected Behavior

无

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

__version__ = "0.2.1"

YilingQiao, 2025-02-20:
We were referring to GPU thread for simulation.

YilingQiao, 2025-02-20:
```
THREADS=$(ps -o nlwp= -p $PID)
```
Is not for this purpose.

closed 2025-02-20 by YilingQiao