Genesis-Embodied-AI/genesis-world #710: [Feature]: Differentiable Physics for Rigid Body Simulation

wangwei1233x, 2025-02-11:
### What feature or enhancement are you proposing?

Hi, just wonder if the project owners have the intention to add differentiable physics for rigid body simulation? Seems like the current differentiable physics framework only works for MPMSolvers... Thanks! 

### Motivation

Wanted to do differentiable physics with rigid objects, and currently the engine is not supporting it.

### Potential Benefit

Expand the application of this engine

### What is the expected outcome of the implementation work?

.

zswang666, 2025-02-12:
This is on the roadmap. Will add in the future. Stay tuned!

wangwei1233x, 2025-02-12:
> This is on the roadmap. Will add in the future. Stay tuned!

Thank you very much for you reply! I'm excited to hear that, and looking forward to the update!

closed 2025-02-12 by wangwei1233x