Genesis-Embodied-AI/genesis-world #716: [Feature]: Function to save/load state of the simulator in/from file

jotix16, 2025-02-12:
### What feature or enhancement are you proposing?

Directly trying to pickle the state fails since the siulation contains Taichi objects (like fields or expressions) that cannot be directly pickled.
``` python
...
state = scene.get_state()

import pickle
# save states
with open("state.pkl", "wb") as f:
    pickle.dump(state, f)
```

throws

``` bash
Traceback (most recent call last):
  File "<string>", line 22, in <module>
TypeError: cannot pickle 'taichi._lib.core.taichi_python.Expr' object
```

### Motivation

Motivation: create dataset

I want to split simulation from rendering. I.e. simulate once and only save the states. And render offline.

### Potential Benefit

This would allow the user to simulate once and render offline.

### What is the expected outcome of the implementation work?

- provide function save_state()
- provide function load_state()

AndreaCastiella, 2025-04-15:
I'm also interested in this!

duburcqa, 2026-08-30:
Reopening: #1315 closed this without answering either half of it.

It does not make `scene.get_state()` picklable, which is what this issue reports. On current `main`, `pickle.dumps(scene)` still raises, and the first obstacle is not a Taichi field at all: `RigidOptions` has no `__pydantic_private__`. A solver and an entity fail the same way.

What it added instead is `Scene.save_checkpoint` / `load_checkpoint`, a pickle of raw field dumps keyed by class and attribute name. That machinery saves a fraction of the state it claims. `Solver.dump_ckpt_to_numpy` reads `self.__dict__` only, so everything a solver's components own is invisible: collider contacts and contact cache, constraint state and warm start, and the coupler's particle attachments. On a scene of a Franka, a plane and three stacked boxes, 247 of 538 arrays are saved. A restored scene therefore carries on from contacts the loading run computed rather than from the ones in the file.

The accompanying test cannot detect that. It builds a Franka alone in free space, drives a position target for 400 steps until it settles, and compares joint positions after restore. A converged target is an attractor, and with no contact geometry the missing arrays hold zeros, so the test passes with the entire collider and constraint state dropped. Adding a plane and a box stack and asserting that the restored scene replays the following steps makes it fail at the first step.

`load_ckpt_from_numpy` also ends each attribute with `if key_base not in arr_dict: continue`, so a file from another scene, a renamed field or a changed shape restores partially and says nothing.

The plan is to remove `save_checkpoint` and `load_checkpoint` rather than repair them, and to answer this issue properly afterwards: a picklable object graph first, then a versioned artifact that a user can attach to a bug report and a maintainer can load without the original assets.

duburcqa, 2026-11-12:
This feature has been implemented.

PR #1315 (2025-06-27): [FEATURE] Scene saver
- A simple file-based checkpoint system.
- Helpers gs.states.save_ckpt(sim, path) and gs.states.load_ckpt(sim, path) restore the full simulator state to a compressed NPZ and meta file. 
- Each solver now exposes dump_ckpt_to_numpy and load_ckpt_from_numpy. Scalar fields become ndarrays, vector / matrix fields a 1-elem object array.
- Resolves #716
closed 2026-11-12 by duburcqa