Genesis-Embodied-AI/genesis-world #717: [Feature]: Keep contact info on GPU

alexanderswerdlow, 2025-02-12:
### What feature or enhancement are you proposing?

Keep contact into on GPU (return as a tensor vs. numpy array). This results ~5x the FPS compared to the same numpy code.

In the plot below, orange is no constraint checking, purple is with the code below, and green is with the original code.

[image]

Draft code (in `genesis/engine/entities/rigid_entity/rigid_entity.py`):
```
    @gs.assert_built
    def get_contacts(self, with_entity=None):
        # Assume contact_data and n_contacts are already torch tensors.
        scene_contact_info = self._solver.collider.contact_data
        n_contacts = self._solver.collider.n_contacts

        valid_mask = torch.logical_or(
            torch.logical_and(
                scene_contact_info.geom_a.to_torch(device='cuda') >= self.geom_start,
                scene_contact_info.geom_a.to_torch(device='cuda') < self.geom_end,
            ),
            torch.logical_and(
                scene_contact_info.geom_b.to_torch(device='cuda') >= self.geom_start,
                scene_contact_info.geom_b.to_torch(device='cuda') < self.geom_end,
            ),
        )

        if with_entity is not None:
            if self.idx == with_entity.idx:
                gs.raise_exception("`with_entity` cannot be the same as the caller entity.")
            valid_mask = torch.logical_and(
                valid_mask,
                torch.logical_or(
                    torch.logical_and(
                        scene_contact_info.geom_a.to_torch(device='cuda') >= with_entity.geom_start,
                        scene_contact_info.geom_a.to_torch(device='cuda') < with_entity.geom_end,
                    ),
                    torch.logical_and(
                        scene_contact_info.geom_b.to_torch(device='cuda') >= with_entity.geom_start,
                        scene_contact_info.geom_b.to_torch(device='cuda') < with_entity.geom_end,
                    ),
                ),
            )

        # Create an index tensor for contacts and compare against per-env n_contacts.
        # Assumes valid_mask.shape[0] corresponds to the maximum number of contacts.
        contact_indices = torch.arange(valid_mask.shape[0], device=valid_mask.device).unsqueeze(1)
        valid_mask = torch.logical_and(valid_mask, contact_indices < n_contacts.to_torch(device='cuda'))

        if self._solver.n_envs == 0:
            valid_idx = torch.nonzero(valid_mask.squeeze(), as_tuple=True)[0]
            contact_info = {
                "geom_a": scene_contact_info.geom_a.to_torch(device='cuda')[valid_idx, 0],
                "geom_b": scene_contact_info.geom_b.to_torch(device='cuda')[valid_idx, 0],
                "link_a": scene_contact_info.link_a.to_torch(device='cuda')[valid_idx, 0],
                "link_b": scene_contact_info.link_b.to_torch(device='cuda')[valid_idx, 0],
                "position": scene_contact_info.pos.to_torch(device='cuda')[valid_idx, 0],
                "force_a": -scene_contact_info.force.to_torch(device='cuda')[valid_idx, 0],
                "force_b": scene_contact_info.force.to_torch(device='cuda')[valid_idx, 0],
            }
        else:
            max_env_collisions = int(torch.max(n_contacts.to_torch(device='cuda')).item())
            contact_info = {
                "geom_a": scene_contact_info.geom_a.to_torch(device='cuda')[:max_env_collisions].t(),
                "geom_b": scene_contact_info.geom_b.to_torch(device='cuda')[:max_env_collisions].t(),
                "link_a": scene_contact_info.link_a.to_torch(device='cuda')[:max_env_collisions].t(),
                "link_b": scene_contact_info.link_b.to_torch(device='cuda')[:max_env_collisions].t(),
                "position": scene_contact_info.pos.to_torch(device='cuda')[:max_env_collisions].permute(1, 0, 2),
                "force_a": -scene_contact_info.force.to_torch(device='cuda')[:max_env_collisions].permute(1, 0, 2),
                "force_b": scene_contact_info.force.to_torch(device='cuda')[:max_env_collisions].permute(1, 0, 2),
                "valid_mask": valid_mask[:max_env_collisions].t(),
            }
        return contact_info
```

### Motivation

Increased speed.

### Potential Benefit

Increased speed.

### What is the expected outcome of the implementation work?

Increased speed.

Kashu7100, 2025-02-13:
Hi, Alex, can you tell me how you benchmarked this?

coghks625, 2025-02-13:
I think the issue I experienced in #687   might also be resolved using this approach.

PR #724 (2025-02-13): [MISC] torch contact

## Description

## Related Issue

Resolves #717 

## Motivation and Context

## Ho
closed 2025-02-13 by Kashu7100