Genesis-Embodied-AI/genesis-world #718: [Bug]: ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.

zanxueyan, 2025-02-13:
### Bug Description

当我执行官方提供的范例代码
`import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()`
是，运行到
`scene.build()`
时报错
[Genesis] [08:45:36] [INFO] Building scene <f4685c2>...
[W 02/13/25 08:45:36.926 23393] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 02/13/25 08:45:37.216 23393] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 02/13/25 08:45:39.704 23393] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [08:45:39] [INFO] Compiling simulation kernels...
[Genesis] [08:45:44] [INFO] Building visualizer...

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
iimport genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```

[Genesis] [08:45:36] [INFO] Building scene <f4685c2>...
[W 02/13/25 08:45:36.926 23393] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 02/13/25 08:45:37.216 23393] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 02/13/25 08:45:39.704 23393] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [08:45:39] [INFO] Compiling simulation kernels...
[Genesis] [08:45:44] [INFO] Building visualizer...

### Expected Behavior

[Genesis] [09:01:40] [INFO] Building scene <1eda608>...
[W 02/13/25 09:01:40.381 24097] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 02/13/25 09:01:40.799 24097] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 02/13/25 09:01:43.389 24097] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [09:01:43] [INFO] Compiling simulation kernels...
[Genesis] [09:01:47] [INFO] Building visualizer...
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/zxy/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1161, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/zxy/.local/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/home/zxy/.local/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/zxy/.local/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/zxy/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 955, in _time_event
    self.on_draw()
  File "/home/zxy/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._render()
  File "/home/zxy/.local/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1098, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/home/zxy/.local/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/zxy/.local/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/zxy/.local/lib/python3.10/site-packages/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/zxy/.local/lib/python3.10/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
  File "/home/zxy/.local/lib/python3.10/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 465, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/zxy/.local/lib/python3.10/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
  File "/home/zxy/.local/lib/python3.10/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 22.04,]
- GPU/CPU [CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

e7f98d9360e6ab0d60f6cffcb902ae583f31b72c

bxtbold, 2025-02-14:
You can refer to this https://github.com/Genesis-Embodied-AI/Genesis/issues/352#issuecomment-2564272924

duburcqa, 2025-03-04:
The actual issue (which is not related with Taichi warning "The execution order is not guaranteed.") has been fixed.

closed 2025-03-04 by duburcqa