Genesis-Embodied-AI/genesis-world #720: [Bug]: The efficiency of genesis is lower than mjx when collisions increase

wanghanhan, 2025-02-13:
### Bug Description

In an experiment, when n*n*n cube stacks with side length 1 and spacing of 0.01 in the xyz direction fall from height 2, I find the following:
1. With the increase of n, the time of each iteration of genesis increases rapidly (mjx is much more stable).
2. The simulation results of genesis using cg and newton are greatly different from those of mjx simulation.
3. When n>3, the time of each iteration of genesis is greater than that of mjx field.

### Steps to Reproduce

The following is my scene settings:
```python
scene = gs.Scene(
    show_viewer=args.v,
    viewer_options=gs.options.ViewerOptions(
            camera_pos=(35, 0.0, 25),
            camera_lookat=(0.0, 0.0, 0.0),
            camera_fov=40,
        ),
    rigid_options=gs.options.RigidOptions(
        dt=0.01,
        constraint_solver=gs.constraint_solver.Newton,
        tolerance=1e-8, # to match mjx
        use_contact_island=True,
        use_hibernation=True,
    ),
)

plane = scene.add_entity(
    gs.morphs.Plane(),
)

n = 3
size = 1
gap=0.01
height=2
for i in range(n):
    for k in range(n):
        for j in range(n):
            scene.add_entity(
            gs.morphs.Box(
                pos=(i*(1+gap),k*(1+gap),j*(1+gap) + height),
                size=(size, size, size),
            ),
        )
```

### Expected Behavior

1. The time consumed by each step is stable regardless of the value of n.
2. Using different constraint solvers produces the same result.
3. The genesis is more efficient than that of mjx.

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

version number: 0.2.1

YilingQiao, 2025-02-14:
What's the performance of MJX in your test environment? The gap between the boxes is 0.01, while each box's size is 1. The boxes heavily overlapped.

The contact_island and hibernation features are designed for sparse collisions. For example, when simulating a robot in a living room with chairs, tables, boxes, etc., many items form isolated islands. The system finds connected components and resolves collisions within each component.

In this case, where all the boxes are stacked very densely, there is only one island. Therefore, enabling use_hibernation and use_contact_island will only increase overhead.

wanghanhan, 2025-02-14:
The following are my test results (simulation steps = 400):
1. n=1
    Genesis simulation time (hibernation: true; contact_island: true): (1) Newton: 138 ms; (2) CG: 135 ms
    Genesis simulation time (hibernation: false; contact_island: false): (1) Newton: 134 ms; (2) CG: 135 ms
    MJX simulation time: (1) Newton: 5400 ms; (2) CG: 5560 ms
3. n=2
    Genesis simulation time (hibernation: true; contact_island: true): (1) Newton: 456 ms; (2) CG: 413 ms
    Genesis simulation time (hibernation: false; contact_island: false): (1) Newton: 906 ms; (2) CG: 1388 ms
    MJX simulation time: (1) Newton: 9342 ms; (2) CG: 8831 ms
5. n=3
    Genesis simulation time (hibernation: true; contact_island: true): (1) Newton: 14104 ms; (2) CG: 11451 ms
    Genesis simulation time (hibernation: false; contact_island: false): (1) Newton: 10685 ms; (2) CG: 25480 ms
    MJX simulation time: (1) Newton: 11377 ms; (2) CG: 13064 ms
7. n=4
    Genesis simulation time (hibernation: true; contact_island: true): (1) Newton: 86006 ms; (2) CG: 32887 ms
    Genesis simulation time (hibernation: false; contact_island: false): (1) Newton: 95208 ms; (2) CG: 109850 ms
    MJX simulation time: (1) Newton: 15372 ms; (2) CG: 14117 ms

The most strange thing is that I obtained totally different simulation result when i use different constraint solvers and different configurations in genesis

Kashu7100, 2025-02-14:
Could you share your eval script for MJX as well (and Genesis' if different or updated from your initial one)?

wanghanhan, 2025-02-17:
For MJX, I only changed line 50-58 in the script https://github.com/zhouxian/genesis-speed-benchmark/blob/main/multi_obj/test_mjx.py:
```python
size = 1
gap=0.01
height=2
cnt = 0
n=4
for i in range(n):
    for k in range(n):
        for j in range(n):
            pos = np.array([i*(1+gap),k*(1+gap),j*(1+gap) + height])
            xml += f'''
            <body name="obj_{cnt}" pos="{pos[0]:.3f} {pos[1]:.3f} {pos[2]:.3f}">
                <freejoint/>
                <geom name="geo_{cnt}" type="box" size="0.5 0.5 0.5"  contype="1" conaffinity="1" mass="0.01"/>
            </body>
            '''
            cnt+=1
```

And the Genesis's script is the same as the initial one (only change the rigid_options and the value of n for behavior verification).

Despite the difference between MJX and Genesis, what i curious most is the strange behavior of Genesis in this situation (different simulation results with different parameters).

Genesis-Embodied-AI, 2025-02-25:
hibernation is not fully released yet. Could you try hibernation off and contact island true?

duburcqa, 2025-03-23:
You should try again on main branch ? The stability of the simulation for this specific scenario has been improved. Do you have a video showing the results of these "vastly different" simulation results ?

closed 2025-05-24 by duburcqa