Genesis-Embodied-AI/genesis-world #721: [Feature]: the transformer of camera 2 world

gxy-TJU, 2025-02-13:
### What feature or enhancement are you proposing?

When I set up one camera with code like this
`cam_0 = scene.add_camera(
        res=(1280, 960),
        pos=(3.5, 0.0, 2.5),
        lookat=(0, 0, 0.5),
        fov=30,
        GUI=True,
    )`
Can I get the transformer between camera pixel and world ?
Does there are any api I can use?
Thanks for any response!

### Motivation

As a [e.g. civil engineering researcher working on flood simulation]
I want to [e.g. be able to simulate city-scale fluid / terrain simulations]
Because [e.g. this enables the efficient design and validation of flood protection structures].

### Potential Benefit

a Convenient features

### What is the expected outcome of the implementation work?

- [ ]

bxtbold, 2025-02-14:
Can you elaborate your question again? Do you mean you want to access to transformation between camera and world frames?

gxy-TJU, 2025-02-15:
> Can you elaborate your question again? Do you mean you want to access to transformation between camera and world frames?

Yes, when I set up a camera, I want to get the transformer between camera pixel and world coordinate。

bxtbold, 2025-02-17:
Perhaps, what you are looking for is [`Camera.transform`](https://github.com/Genesis-Embodied-AI/Genesis/blob/bf3bc101630bea36b1835f9d9cbbbf8e65596760/genesis/vis/camera.py#L562) which gives a 4x4 transformation matrix.

You can access from your camera instance like this:
```bash
cam_0 = scene.add_camera( res=(1280, 960), pos=(3.5, 0.0, 2.5), lookat=(0, 0, 0.5), fov=30, GUI=True, )
cam_0.transform
```

wangyian-me, 2025-03-20:
You can refer to these code: [link1](https://github.com/wangyian-me/architect_official_code/blob/a743b73961e8ce40b4d1f208a433c8b80994e5f2/genesis_utils/place_scene.py#L553) and [link2](https://github.com/wangyian-me/architect_official_code/blob/a743b73961e8ce40b4d1f208a433c8b80994e5f2/ram_utils/utils_dino15.py#L60).

Also, I will soon update these functions to APIs of Genesis.

wangyian-me, 2025-03-20:
Maybe refer to pull request #897

closed 2025-09-27 by duburcqa