Genesis-Embodied-AI/genesis-world #725: [Feature]: `robot.set_qpos` and `robot.control_dofs_position`  have different indexing

mhwang003, 2025-02-13:
### What feature or enhancement are you proposing?

HI, Thanks for the incredible work!

Using Genesis has been exciting and quite intuitive.

I noticed when using `robot.set_qpos` and `robot.control_dofs_position` for the go2 robot, the indexing for the joints was different. 
for `robot.set_qpos` the indexing was

`FL_indices = torch.tensor([0, 4, 8])
        FR_indices = torch.tensor([1, 5, 9])
        RL_indices = torch.tensor([2, 6, 10])
        RR_indices = torch.tensor([3, 7, 11])` 
and for  `robot.control_dofs_position`  it was

 `FR_indices_target = torch.tensor([0, 1, 2])
        FL_indices_target = torch.tensor([3, 4, 5])
        RR_indices_target = torch.tensor([6, 7, 8])
        RL_indices_target = torch.tensor([9, 10, 11])`

Am I doing something wrong? or If possible having the same indexing would be awesome. Thanks

### Motivation

easier indexing 

### Potential Benefit

less bugs 

### What is the expected outcome of the implementation work?

same indexing for the 2 methods

zswang666, 2025-02-14:
`qpos` and `dofs` are different concepts and thus they won't necessarily share the same indexing. Specifically, for free joints, `qpos` has 7 dimensions and `dofs` has 6. But the actual indexing you provided looks quite weird indeed. Could you provide a simple script that show how you arrived at your observation?

Kashu7100, 2025-02-28:
closing for inactivity, but feel free to reopen if needed.

closed 2025-02-28 by duburcqa